openhwfoundation/cva6 #214: Question in Synthesis by Design Compiler

24Kcool, 2019-04-09:
Hello,

I am a fresh man in ariane project. I meet some questions about syntax when I want to synthesize this project. I use Desigon Compiler (dc_shell) to achieve this movement while it shows me that there are some problems. Can you give some suggestions for them? By the way, the version of dc_shell is 2014. 

Here are the error issue codes:
`analyze -f sverilog ../ariane/src/load_store_unit.sv`
`Running PRESTO HDLC`
`Compiling source file ../ariane/src/load_store_unit.sv`
`Opening include file ../ariane/src/load_unit.sv`
`Opening include file ../ariane/src/store_unit.sv`
`Error:  ../ariane/src/load_unit.sv:303: The construct 'inside operator in a continuous assignment' is not supported.  (VER-721)`
`Error:  ../ariane/src/load_unit.sv:304: The construct 'inside operator in a continuous assignment' is not supported.  (VER-721)`
`Error:  ../ariane/src/load_unit.sv:305: The construct 'inside operator in a continuous assignment' is not supported.  (VER-721)`
`Error:  Cannot recover from previous errors. (VER-518)`
`*** Presto compilation terminated with 4 errors. ***`
`0`
`analyze -f sverilog ../ariane/src/alu.sv`
`Running PRESTO HDLC`
`Compiling source file ../ariane/src/alu.sv`
`Error:  ../ariane/src/alu.sv:137: Syntax error at or near token 'assign'. (VER-294)`
`Error:  ../ariane/src/alu.sv:140: Syntax error at or near token 'assign'. (VER-294)`
`Error:  Too many errors; can't continue. (VER-40)`
`*** Presto compilation terminated with 3 errors. ***`
`0`
`analyze -f sverilog ../ariane/src/mult.sv`
`Running PRESTO HDLC`
`Compiling source file ../ariane/src/mult.sv`
`Error:  ../ariane/src/mult.sv:26: The construct 'inside operator in a continuous assignment' is not supported.  (VER-721)`
`Error:  ../ariane/src/mult.sv:27: The construct 'inside operator in a continuous assignment' is not supported.  (VER-721)`
`Error:  ../ariane/src/mult.sv:68: The construct 'inside operator in a continuous assignment' is not supported.  (VER-721)`
`Error:  ../ariane/src/mult.sv:70: The construct 'inside operator in a continuous assignment' is not supported.  (VER-721)`
`*** Presto compilation terminated with 4 errors. ***`
`0`
`analyze -f sverilog ../ariane/src/load_unit.sv`
`Running PRESTO HDLC`
`Compiling source file ../ariane/src/load_unit.sv`
`Error:  ../ariane/src/load_unit.sv:303: The construct 'inside operator in a continuous assignment' is not supported.  (VER-721)`
`Error:  ../ariane/src/load_unit.sv:304: The construct 'inside operator in a continuous assignment' is not supported.  (VER-721)`
`Error:  ../ariane/src/load_unit.sv:305: The construct 'inside operator in a continuous assignment' is not supported.  (VER-721)`
`*** Presto compilation terminated with 3 errors. ***
`

Is it possible that this project acutally should not be synthesized by dc_shell?
Thank you for any reply for above question.

zarubaf, 2019-04-09:
Hi, it seems that the DC version you use is a bit too old and doesn’t recognize the inside keyword. We usually use a 2017 and later version without problems.

zarubaf, 2019-04-16:
Did you resolve your issue?

24Kcool, 2019-04-17:
We are still trying to solve errors, but it seems that it could slove some problems after analyzing some files.

closed 2019-04-17 by 24Kcool

Shadi1987, 2022-03-14:
hello,

I'm facing the same problem with "inside". Even after analyzing the other files the problem of "inside" appears.

zarubaf, 2022-03-15:
Upgrading DC should be the solution :-)