openhwfoundation/cva6 #226: Will the Vivado do the retiming successfully for FPU with a 100MHz target frequency?

hyf6661669, 2019-04-25:
At first I must say your open-source FPU is really helpful for me, and I learn lots of knowledge about how to design FPU from it.

I find that in ```fpnew_pipe_out.sv``` you just add some registers after the output. It also says:
```
// NOTE: These registers must be retimed in synthesis for sensible pipelining. Make sure to
// optimize registers through the instantiating hierarchy.
```

So it seems that you just let the synthesis tools to do the retiming work, instead of doing it manually. Have you found out the max frequency that ariane with FPU could run on a GENESYS2 board (with XC7K325T-2)? I am wondering whether Vivado (my Vivado version is 2018.2) could do the retiming work successfully, with a 100MHz target frequency. Unfortunately my computer (I3-8100, 8GB DDR3) is really slow. It has been running implementation for at least 2.5 hours and still can't give a result.

I'm really appreciated if anyone could give any information.

msfschaffner, 2019-04-25:
Hi @hyf6661669, 

Thanks for your interest in the Ariane project! 

Yes, we currently rely on the tools to properly balance these registers. On ASIC flows this seems to work fine, but so far we could not really get it working well with the Vivado FPGA flow.

Just FYI, we are actively working on an improved version with hand placed registers which should improve timing significantly on FPGA mappings. You can expect that the complete mapping runs around 60MHz on a Kintex-7 (Genesys2). 100MHz is a bit too tight, even without FPU.

This improved version will be upstreamed very soon (hopefully this week).

Best,
Michael

hyf6661669, 2019-04-25:
@msfschaffner 
Thanks. Do you mean that the improved version of ariane is expected to run at about 60MHz on Genesys2 with FPU enabled?

msfschaffner, 2019-04-26:
Yes. We currently use 60 MHz for the Kintex7 FPGA mappings (both the standalone version from this repository and on OpenPiton). The FPU fix will allow us to use the FPU at that frequency as well (previously we had to lower to 30MHz with FPU enabled).

hyf6661669, 2019-04-26:
@msfschaffner 
That sounds amazing! I'm looking forward to studying your latest version!

closed 2019-04-26 by hyf6661669

mysoreanoop, 2021-07-19:
Hi Michael, 
Sorry, this is a little off topic, but were you able to figure out why exactly Vivado wasn't doing the retiming as expected? Some posts mention that certain older Vivado versions weren't smart enough to do backwards retiming when multiple nets were converging on the register; any chance this was the reason with your FPU? Also, if you have tried the latest Vivado versions, do you see any improvement in retiming?

zarubaf, 2021-07-19:
In my experience, the retiming in Vivado is still quite weak compared to Synopsys. The FPU now has a version with naïvely distributed registers which should perform a bit better on the FPGA. 100 MHz should be possible on Ultrascale+ devices.