openhwfoundation/cva6 #230: The FPU seems not fully IEEE 754-2008 compliant

hyf6661669, 2019-04-27:
[image]

From my understanding, when a floating-point operation gets a +inf/-inf result, the value that should be stored in floating-point register (fpr) is influenced by the rounding mode. Let's think about single-precision floating-point number. If the result is +inf and the rounding mode is ```roundTowardNegative``` or ```roundTowardZero```, then we should write 0x7f7f_ffff into a fpr. If the result is -inf and the rounding mode is ```roundTowardPositive``` or ```roundTowardZero```, then we should write 0xff7f_ffff into a fpr. 

However, in fpnew_fma.sv, when we get a +inf/-inf result, you just set ```special_result``` to 0x7f80_0000 or 0xff80_0000:
[image]

It seems that the FPU doesn't fully follow the IEEE 754-2008. The same issue might also exist in other FPU modules, as I have only read the source codes of the FMA module.

Is it a must for us to design a hardware FPU that fully follows the IEEE 754-2008?

stmach, 2019-04-27:
Hi, thanks for using our FPU!

You are correct that FP operations that overflow decide according to rounding mode whether +-inf or +-MAXFLOAT is the result. 
However, the code you are referencing deals with the case where the inputs of the operation are already +-infinity. in such cases, the (infinitely precise) result of the operation is also +-infinity, and no rounding mode can change this. E.g. take +infinity - (any finite number). The result is still +infinity due to mathematical definition.

The regular case where an operation overflows is handled in the regular data path, by setting the result to +-MAXFLOAT (plus round and sticky bits) before rounding which then rounds according to the IEEE754-2008 standard. Thus the operations that naturally overflow are rounded correctly.

hyf6661669, 2019-04-27:
@stmach 
My previous understanding about +inf/-inf was wrong, thanks for your explanation.

closed 2019-04-27 by hyf6661669