openhwfoundation/cva6 #233: FPU single precision add inaccuracy

jrrk, 2019-05-02:
I am testing the improved, pipelined FPU under ariane-dev branch. I am trying to run TestFloat and in the main it works well, but I am having problems with certain particular operands, as per the attached testcase:

  TEST_FP_OP2_S_( 2, fadd.s,             33354290,             33357822,   -3531
.952392578125 ); #   actually returns  33357822
  TEST_FP_OP2_S_( 3, fadd.s, -1.701411733192644E+38, -1.701411733192644E+38, -2.
350970765024232E-38 ); #  actually returns NAN

[rv64uf-p-custom.zip](https://github.com/pulp-platform/ariane/files/3137711/rv64uf-p-custom.zip)

I can add more test vectors later if needed

msfschaffner, 2019-05-02:
Hi Jonathan, thanks for testing! I'm forwarding this one to @stmach who designed the FPU. Steve, could you take a look?

jrrk, 2019-05-02:
Postscript: I have found that the FPGA does not always produce consistent results. Perhaps there is a problem with saving and restoring the FPU state during a context switch ?

jrrk, 2019-05-03:
We have a workaround [here](https://github.com/lowRISC/linux/pull/1)

zarubaf, 2019-05-06:
Thanks for bisecting this bug.  I will go through the state dirtying code to see which instructions might be missing.

zarubaf, 2019-05-14:
That is indeed stupid and a mistake. Instead of the `0` it should be `i`.

jrrk, 2019-05-14:
Unfortunately that doesn't seem to help either. Perhaps there are some calculations in progress that don't get saved when the co-processor status is saved ? Perhaps you could suggest some signals to look at in the internal logic analyzer ?

jrrk, 2019-05-14:
@zarubaf I think I found the problem, see associated pull request [here](https://github.com/pulp-platform/ariane/pull/240)

PR #238 (2019-05-14): fp_dirty: Dirty on fp register write
That should fix loads to floating point regs not dirtying the FP state (should fix #233). 

@jrrk I'd greatly appreciate if you could test the fix with your setup.
closed 2019-05-14 by jrrk