openhwfoundation/cva6 #237: Golden reference model for the Ariane

Bhagyaraja, 2019-05-12:
Hello,

We wanted to do verification of the Ariane SoC. For this verification, we need some reference model of the Ariane SoC. Do you have any reference model for the Ariane SoC or reference model of the processor core?  
Kindly point us the reference architecture/ c-model if you have any of these? 

Thanks,
Bhagyaraja Adapa

msfschaffner, 2019-05-13:
Hi @Bhagyaraja 

Our current verification environment for the core includes the RISC-V assembly tests, benchmarks and RISC-V torture (see main Makefile in the repository). Further, we have been using Spike as a reference model for the core (Spike is the RISC-V ISA simulator: https://github.com/riscv/riscv-isa-sim). @zarubaf provisioned a sort of "tandem" simulation where Spike is used to double check the commited instructions. This works well up to the point where interrupts and timers are involved, since Ariane may show a different timing behaviour in that case due to pipeline latencies. 

We do not have a golden model for the SoC, but it should be possible to put together a similar SoC configuration in QEMU since it has RISC-V support (https://wiki.qemu.org/Documentation/Platforms/RISCV).

Best,
Michael

raulbehl, 2019-05-13:
Hi MIchael/Florian, 

Interested to know a bit more about the following point: 

> @zarubaf provisioned a sort of "tandem" simulation where Spike is used to double check the commited instructions.

What I remember is that there was a log comparison done between Ariane and Spike sims by running the same the test on RTL and simulator. Is it the same setup or something different?

msfschaffner, 2019-05-13:
Yes that is the checker that I was referring to.

zarubaf, 2019-05-13:
No, actually it is really Spike compiled to a library and DPI called from Verilog.

closed 2019-10-09 by zarubaf

sago5993, 2019-10-23:
@zarubaf Can you point me to opensource checker code where spike is compiled to library and DPI called from verilog