openhwfoundation/cva6 #242: Divider Rounding Issues

zarubaf, 2019-05-15:
> There is still work to do on sub-normalised divides but this is repeatable and unrelated. For example 2.350988561514729E-38 / 2 => 0. (it should give 1.175494350822288E-38, a subnormal value)

Hello, thanks for your inputs.
Indeed we are aware of bugs in the FP divider and square-root unit, however this one I haven't seen yet (see end of post for more thoughts on the bug).
The div/sqrt unit we currently have is not very configurable and the code is quite convoluted, so we are hoping to replace it with something more flexible (and IEEE compliant) soon. Hence a bugfix for the current div/sqrt is not a high priority at this time.

Long-winded addendum
The result value you list is correct (under default rounding), however it's actually encoded as the smallest normal FP32 value 0x00800000. The infinitely precise result lies exactly between this result and the largest subnormal FP32 value.
The div/sqrt bugs we've seen so far concern off-by-one-ulp because of wrong rounding. What's possibly happening here seems to be that the mantissa is correctly rounded from all-ones (from the infinitely precise, subnormal result) up to all-zeroes, however the overflow produced from rounding doesn't seem to be added onto the (subnormal = 0) exponent. This leads to the zero result.
I don't know whether this bug is contained to this subnormal/normal border case, but it might be possible that mantissa overflows from rounding in general are buggy.

ivanvera84, 2022-02-22:
Hi, I've run Tortoise tests on Ariane and there are some mismatches with the signature produced by Spike regarding the result from the FPU (looks like a rounding issue in sqrt). 
I was to open an issue when I see this one. 
Is there any update or is it planning any action?

JeanRochCoulon, 2023-04-16:
Hello @ivanvera84    The FPU has just been updated. Can you verify whether the bug has disappeared, in that case the github issue could be closed.