openhwfoundation/cva6 #260: How to preload the RISC-V ELF file when running verilator simulation?

hyf6661669, 2019-06-05:
I find that using the debug module to load the RISC-V ELF files is really slow. The simulation would take lots of time to load ```riscv-pk``` and user codes/data into the DRAM.  It would be very helpful if you can provide a method for preloading in verilator simulation. In https://www.mentor.com/products/fpga/download/modelsim-pe-simulator-download.
It's said that the latest verison of ModelSim is not useable in Linux.
[image]

Much thanks!

zarubaf, 2019-06-14:
Thanks for giving us feedback. Unfortunately, this kind of feature is not very high on our todo list currently as we do not use Verilator for development. I am absolutely positive that ModelSim greater than 10.5 run on Linux and if you want to preload this is the preferred way of doing so.

MikeOpenHWGroup, 2023-02-18:
Hi @hyf6661669, thanks for your interest in CVA6.  This issue is truly ancient and by now CVA6 has changed a lot and Verilator is well supported.  I will close this issue.

closed 2023-02-18 by MikeOpenHWGroup