openhwfoundation/cva6 #261: MEIP

suppamax, 2019-06-14:
Hi,

in your CSR implementation, why is MIP_MEIP not in the mask for the MIE register?

Thanks,
Max

zarubaf, 2019-06-14:
From Privileged Spec 3.1.0: The `MEIP` field in `mip` is a read-only bit that indicates a machine-mode external interrupt is pending. `MEIP` is set and cleared by a platform-specific interrupt controller.

suppamax, 2019-06-14:
Sorry, my statement was unclear.

in csr_registers I see the following
`                riscv::CSR_MIE: begin
                    mask = riscv::MIP_SSIP | riscv::MIP_STIP | riscv::MIP_SEIP | riscv::MIP_MSIP | riscv::MIP_MTIP;
                    mie_d = (mie_q & ~mask) | (csr_wdata & mask); // we only support supervisor and M-mode interrupts
                end
`
So MIE.MEIE is not writable ("| riscv::MIP_MEIP" is missing when setting mask), and as a consequence PLIC-generated interrupts are not allowed in M-mode.
Is this correct?

zarubaf, 2019-07-10:
Yes you are potentially right I think.

axel-h, 2019-07-10:
I wonder, Is this the same as https://github.com/pulp-platform/ariane/issues/267

commit b9990f6f (2019-08-12): csr_regfile: Fix #261 non-setable MEIE bit
commit a147562c (2019-08-19): csr_regfile: Fix #261 non-setable MEIE bit (#302)
closed 2019-08-19 by zarubaf