openhwfoundation/cva6 #263: Deliver timer interrupts to lower privilege levels

suppamax, 2019-06-19:
Hi,

from the specs I see that it should be possible to deliver timer interrupt from M-mode to lower privilege modes by means of MIP.STIP and MIP.UTIP.
But from the csr_regfile implementation I see that it is possible to set trap_to_priv_lvl to values different from M only by means of delegation.

Is my understanding right? 
What's the reason for this choice?
Is this a violation of the specs?

Thanks,
Max

zarubaf, 2019-06-19:
Traps are always taken in M-mode unless delegated.

suppamax, 2019-06-19:
I see.

But then I don't understand this statement in the RISCV specs: "The UTIP and STIP bits may be written
by M-mode software to deliver timer interrupts to lower privilege levels".
How did you interpret it?

What I had expected was:
- software running in U-mode
- timer_irq
- M-mode trap handler:
        - mie.mtip=0
        - mip.stip=1
        - mret
=> here I expect the S-mode interrupt handler to start, because of STIP
What happens instead is
- U-mode continues
- M-mode trap handler started because of mip.stip

zarubaf, 2019-06-24:
You need to delegate the S-Mode timer interrupt to S-Mode.

zarubaf, 2019-07-10:
Does that answer your question? If so, please close this issue.

suppamax, 2019-07-10:
I interpreted this sentence 
"The UTIP and STIP bits may be written by M-mode software to deliver timer interrupts to lower privilege levels"
as "independently from delegation".
But if you say that delegation is anyway required to deliver interrupts to S-mode, fine for me.

closed 2019-07-10 by suppamax

zarubaf, 2019-07-10:
Yes 100% sure. Somewhere the manual says: **All** interrupts and exceptions are taken in M-Mode except if they are delegated with `[s|m]delegate`. The only difference with `UTIP` and `STIP` are that they _can_ be delegated.

berendo, 2021-10-21:
@zarubaf I'm coming late to this thread after having read the historical discussion in https://groups.google.com/a/groups.riscv.org/g/isa-dev/c/iJqwTw2hIRk and then stumbling onto this issue. The idea of a software-write to the `STIP` being involved in what's otherwise a hardware delegation method seems a bit counter-intuitive. Furthermore, if the M-mode exception handler first has to set `mie.mtip` to `0` before `mret`'ing to immediately enter S-mode for delegation, once the S-mode timer interrupt handler is done it has to re-escalate to M-mode to set `mie.mtip` back to `1`.

For pure hardware delegation, shouldn't the incoming interrupts also be factored into `mip_d` under the `riscv::IRQ_S_*` positions [here](https://github.com/openhwgroup/cva6/blob/5c5c704d1d14190ec349fae12fe39fbe025456ae/core/csr_regfile.sv#L647-L655) as well?

I will go poke into the QEMU RISC-V model as well...