openhwfoundation/cva6 #265: Ariane sb_full_o permanently high

jrrk, 2019-06-20:
Dear Florian,

Great conference last week, nice to meet you at last

I have got a bit further with Ariane booting using systemd. It turns out that the serial console needs a longer timeout than the default 30 seconds, then it will be recognised. Now the Ariane reaches the login prompt, but just gets stuck with sb_full_o (scoreboard full?) permanently high. I am a little bit unclear what this means, if you have suggestions for other signals to monitor I will be pleased to hear from you (I am using ILA in the FPGA environment, not simulation).

I am using tagged release 4.2.0

Regards,

Jonathan

[iladata.zip](https://github.com/pulp-platform/ariane/files/3310197/iladata.zip)

jrrk, 2019-06-20:
[iladata.zip](https://github.com/pulp-platform/ariane/files/3310773/iladata.zip)

jrrk, 2019-06-20:
Another trace (with expanded scoreboard) appears to suggest a complete lock up when 

fdiv.d  fs1, fa5, fs1
is followed by:
fdiv.d  fs3, fa5, fs3

[iladata2.zip](https://github.com/pulp-platform/ariane/files/3310961/iladata2.zip)

jrrk, 2019-06-20:
It is tentatively established by swapping the co-processor that the fundamental cause of the lockup is the floating point co-processor.

zarubaf, 2019-06-21:
Thanks Jonathan, was great to meet you too. That is great progress, also on den debugging of the cause. I assume that the floating point divider ist not handling the tag propagation correctly and ultimately causes a hang up. @stmach reported that he had fixed this and it was discovered in one of the torture tests. Recently we deactivated the torture test from CI as it would cause timeouts. Unfortunately, I am traveling right now but hope to be able to look into this the next days. Ideally debugging the torture test would also fix this divider issue. Maybe @stmach can have a look whether the fix he did ist still in the codebase.

jrrk, 2019-07-10:
Hi Florian,
  Any news and/or timescale to review this issue?

Regards,
Jonathan

stmach, 2019-07-10:
Hello,
I managed to reproduce the bug using torture testing, indeed the div/sqrt unit can lose an in-flight operation. It manifested itself when the core tries issuing a new div/sqrt operation exactly 2 cycles after the previous one.
The bug should be fixed in v0.6.1 of the FPU, and I submitted a PR to update the FPU to this version (#277).

zarubaf, 2019-07-14:
The updated FPU should now be in the latest Ariane version. @jrrk Could you re-check if that solves the problem?

jrrk, 2019-07-14:
Will do.

jrrk, 2019-07-16:
The Ariane no longer locks up during booting, and I am able to log onto the console. However when I launch startx the server output never appears. This may be related to the ssh-server random startup delay problem which could both implicate the high-resolution timer support peripheral. The CPU doesn't respond to crtl-C but Ethernet ping remains operational, so definitely not the same behaviour as before. Perhaps that should be opened as a new issue.

zarubaf, 2019-07-25:
Yeah lets move this to a new issue. Thanks for the heads-up.

closed 2019-07-25 by zarubaf