openhwfoundation/cva6 #266: run-asm-tests with modelsim = fatal errors

dsisejkovic, 2019-06-21:
I have been trying to run the asm tests (in batch mode) on the latest ariane version.
However, I run into multiple warnings and a fatal error.

What am I doing wrong?

The command: 

`make run-asm-tests batch-mode=1`

The output:

>\# ** Warning: (vsim-3015) >/net/home/user/cores/har_v4.2/ariane/src/frontend/instr_queue.sv(319): [PCDPC] - Port >size (1) does not match connection size (2) for port 'd_i'. The port definition is at: >/net/home/user/cores/har_v4.2/ariane/src/common_cells/src/unread.sv(17).
>\#    Time: 0 ns  Iteration: 0  Instance: >/ariane_tb/dut/i_ariane/i_frontend/i_instr_queue/i_unread_instr_fifo File: >/net/home/user/cores/har_v4.2/ariane/src/common_cells/src/unread.sv
>\# Loading ./work-dpi/ariane_dpi.so
>\# Loading /net/sw/mentor/modelsim10.4/questasim/uvm-1.1d/linux_x86_64/uvm_dpi.so
>\# coverage save -onexit tmp/.ucdb
>\#  run -a
>\# ----------------------------------------------------------------
>\# UVM-1.1d
>\# (C) 2007-2013 Mentor Graphics Corporation
>\# (C) 2007-2013 Cadence Design Systems, Inc.
>\# (C) 2006-2013 Synopsys, Inc.
>\# (C) 2011-2013 Cypress Semiconductor Corp.
>\# ----------------------------------------------------------------
>\# ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
>\#    Time: 0 ns  Iteration: 0  Instance: >/ariane_tb/dut/i_ariane_peripherals/gen_uart/i_apb_uart/UART_RXFF
>\# ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
>\#    Time: 0 ns  Iteration: 0  Instance: >/ariane_tb/dut/i_ariane_peripherals/gen_uart/i_apb_uart/UART_TXFF
>\# ** Fatal: Idx_o / gnt_o do not match.
>\#    Time: 5570 ns Started: 5570 ns  Scope: >ariane_tb.dut.i_ariane.issue_stage_i.i_scoreboard.i_sel_rs1.genblk1.gnt_idx File: >/net/home/user/cores/har_v4.2/ariane/src/common_cells/src/rr_arb_tree.sv Line: 231
>\# ** Note: $finish    : >/net/home/user/cores/har_v4.2/ariane/src/common_cells/src/rr_arb_tree.sv(231)
>\#    Time: 5570 ns  Iteration: 9  Region: >/ariane_tb/dut/i_ariane/issue_stage_i/i_scoreboard/i_sel_rs1/genblk1/gnt_idx
>\# End time: 15:39:57 on Jun 21,2019, Elapsed time: 0:00:03
>\# Errors: 1, Warnings: 3
>\\# Saving coverage database on exit...
>vsim +permissive -t 1ns -64 -coverage -classdebug   -c -do "coverage save -onexit tmp/.ucdb; >run -a; quit -code [coverage attribute -name TESTSTATUS -concise]" -do " log -r /*; run -all;" >+jtag_rbb_enable=0 -lib work +max-cycles=10000000 +UVM_TESTNAME=core_test \
>+BASEDIR=/net/heap/user-pub/tools/riscv-tools/riscv-tests/build/isa/ >+UVM_NO_RELNOTES +UVM_VERBOSITY=LOW +jtag_rbb_enable=0  -gblso >/net/heap/user-pub/tools/riscv-tools/lib/libfesvr.so -sv_lib work-dpi/ariane_dpi        \
>ariane_tb_optimized  +permissive-off ++/net/heap/user-pub/tools/riscv-tools/riscv->tests/build/isa//rv64ui-p-addi ++ | tee tmp/riscv-asm-tests-rv64ui-p-addi.log>
>Reading pref.tcl

zarubaf, 2019-06-21:
The behavior is strange. My first guess would be your (old) ModelSim installation not understanding the RTL correctly. Especially:
```
# ** Warning: (vsim-3015) >/net/home/user/cores/har_v4.2/ariane/src/frontend/instr_queue.sv(319): [PCDPC] - Port >size (1) does not match connection size (2) for port 'd_i'. The port definition is at: >/net/home/user/cores/har_v4.2/ariane/src/common_cells/src/unread.sv(17).
```
https://github.com/pulp-platform/ariane/blob/e873743b547fbf39bd08f25a0bfa7db38eb61794/src/frontend/instr_queue.sv#L319

Clearly the port size is `1` after or-reduction. As I don't see anything wrong with the RTL (and our internal CI passes with Modelsim 10.6) I'd suggest you re-try with a newer simulator. Sorry for not being mor of help.

dsisejkovic, 2019-06-22:
Yes. You are right. Thanks! With version 10.7 it seems to complete successfully.
I still get two warnings - can this be ignored?

>
> ----------------------------------------------------------------
> ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
>    Time: 0 ns  Iteration: 0  Instance: /ariane_tb/dut/i_ariane_peripherals/gen_uart/i_apb_uart/UART_RXFF
> ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
>    Time: 0 ns  Iteration: 0  Instance: /ariane_tb/dut/i_ariane_peripherals/gen_uart/i_apb_uart/UART_TXFF
> UVM_INFO tb/ariane_tb.sv(113) @ 1930230: reporter [Core Test] *** SUCCESS *** (tohost = 0)
> ** Note: $finish    : tb/ariane_tb.sv(116)
>    Time: 1930230 ns  Iteration: 3  Instance: /ariane_tb
> Saving coverage database on 
> End time: 18:20:16 on Jun 22,2019, Elapsed time: 0:00:30
> Errors: 2, Warnings: 2

zarubaf, 2019-06-22:
Yes, they are spurious VHDL warnings at time zero. Everything should be fine after the reset kicked in. Glad that fixed that for you. Might be adding a clarification in the `README` on the tool version.

closed 2019-06-22 by dsisejkovic