openhwfoundation/cva6 #267: Does ariane support Machine mode external interrupt?

hyf6661669, 2019-06-23:
In https://github.com/pulp-platform/ariane/blob/master/src/csr_regfile.sv#L490-L493:
```

riscv::CSR_MIE: begin
                    mask = riscv::MIP_SSIP | riscv::MIP_STIP | riscv::MIP_SEIP | riscv::MIP_MSIP | riscv::MIP_MTIP;
                    mie_d = (mie_q & ~mask) | (csr_wdata & mask); // we only support supervisor and M-mode interrupts
                end

```

It seems that mie_q[MIP_MEIP] (mie_q[11]) is always set to 0.

hyf6661669, 2019-06-27:
@zarubaf  Could you please have a look?

zarubaf, 2019-07-25:
I think this is indeed a problem. I will have a look at it once I've a bit more time. Sorry for delaying it. Closing it as it is a duplicate issue.

closed 2019-07-25 by zarubaf