openhwfoundation/cva6 #268: Write data to a particular address

sujinyuan, 2019-06-24:
I tried to write data to a particular address(like 0xd0000000) using c, like this:
`volatile uint64_t *loc_addr = (volatile uint64_t *) addr ;`
` *loc_addr = value;`
It worked in the Verilator. But  there was a Segmentation fault when i tried it in the FPGA.
It seems like i don't have access to the address in linux.
Is there a way to write data to a particular address in the FPGA?
Much thanks!

zarubaf, 2019-06-24:
Linux works with virtual addresses, this is nothing specific to Ariane, and you will need to write a device driver if you want to access physical memory.

sujinyuan, 2019-06-25:
Thanks for your reply. I will try it according to your suggestion.

closed 2019-06-25 by sujinyuan