openhwfoundation/cva6 #270: linux developing issue

sujinyuan, 2019-06-25:
Sorry to send issue again. 
We have implemented AXI-based arianne interface with in-house developped AI accelerator. Currently we start testing the AI accelerator through Linux kernel running on arianne. The approach is putting several images into the sd card from host machine and process those in Linux kernel on digilent GENESYS2 FPGA.
But I'm not familiar with the Linux kernel. Here are Several questions:

1.We cannot write to physical address (0xD0000000) for AI module configurations through pointers directly from C file in Linux. There should be a software driver handling this. Is there any example on such driver design?
2. Seems that gcc is missing in the kernel, how to combine and load such driver?
3. The current busybox based Linux kernel seems don't have standard library such as glibc (fopen)? How can we build this into the kernel? Please refer any tutorial in case there is.
4. After booting of kernel, can I locate image files stored in the SD card in arianne? 

Much Thanks!

zarubaf, 2019-06-25:
1. Well, the Linux kernel is full of drivers ;-). You can have a look if you find a driver which is doing similar things. There are also books on Linux device driver development and the internet should also provide a lot of information. 
2.  First of all, `gcc` is not part of the Linux kernel. But it is true that we don't package `gcc` into the `rootfs`. Compilation on a 50 MHz, single-issue, in-order core is not a lot of fun, so I would suggest you cross-compile the module. Then you have two options:
- You compile it directly into the Linux kernel.
- You build a loadable kernel-module and load it from the SD Card or the `rootfs` overlay. You most probably want to activate support for loadable kernel modules if you go down that path.
3. I am sure there is.
4. Yes. You have to mount the second partition of the SD Card (given it is formatted with a file-system the Linux kernel has support for).

sujinyuan, 2019-07-04:
I planned to use mmap(user level) or ioremap(kernel level) to access address.I added a node in device tree(dts)'s soc part: 
`nnacc@d0000000 {`
`    reg = <0x0 0xd0000000 0x0 0x1000000>;`
 `};` 
But it didn't work,seems like mmap can't access a address outside the "memory" node's address(in Ariane's dts, it's 0x80000000~0xbfffffff)
Can mmap or ioremap access all nodes' address or just only "memory" node's address?
Thanks a lot!

jrrk, 2019-07-16:
To add a new peripheral on the AXI bus, you need to add it electrically to a new port on the AXI crossbar, and declare its address range as a crossbar parameter. This usage will be adequate for bare-metal testing. To use under Linux you then need to pass the nnacc resource mentioned above to the driver probe function to ioremap() or its variants, in order to return a virtual address that can work in the driver. Then any data passed from user mode to driver can employ copy_from_user() or copy_to_user() as appropriate. If this seems complicated you may find it useful to add a custom peripheral in spike a.k.a riscv-isa-sim to test your software in a debugging environment.

sujinyuan, 2019-07-22:
Thank you so much for your reply! I’m using this method. I added start and end address to the axi crossbar and used ioremap function.But I found the address I declare(like 0xd0000000 ~ 0xd1000000) cannot be accessed and other addresses (like GPIO)can be when I used the same driver code only with different address.

jrrk, 2019-07-22:
My next debugging step would be to run a smoke test of your peripheral in Verilator (make verilate), to establish if the addresses are getting through to your peripheral. You should also study any synthesis warning messages to see if you have made an error somewhere that is detectable at compile time (plenty of errors are not unfortunately). This will allow you to know if you have a hardware or software problem. I would also redirect all accesses to your peripheral in Linux to a macro or inline function. This will allow you to add printk() debugging statements more easily. Last time I checked virtual memory support had not been added to gdb yet - this limits its applicability to Linux debugging.

sujinyuan, 2019-07-23:
I have already run test in Verilator.It can access address and get test results.But it didn't work on FPGA. And thank you for your suggestion to study synthesis warning messages.

Moschn, 2019-08-07:
Did you rebuild the bootrom for the FPGA with the new device tree? The device tree is hardcoded into the bootloader.

sujinyuan, 2019-08-23:
Sorry for so long to reply. 
I can get result of accelerator now,although there are still some bugs. And I found a problem that I haven't figured  out whether it was caused by my design module or Ariane： the _aw_valid_  is trigged in the same cycle as other signals(like _w_data_ or _w_strb_) in Verilator， but is triggered one cycle ahead of the other signals (_w_data_ or _w_strb_) in FPGA using vivado debug tool.

jrrk, 2019-08-23:
As you probably know, individual channels of AXI bus are delay independent. The Ariane repository takes advantage of this to improve timing on the FPGA by adding extra delay cycles compared with Verilator which uses a minimum, timing independent netlist. In addition the FPGA uses a full DDR interface IP whereas Verilator uses a behavioural memory for performance and simplicity.

For this reason the ILA trace will look somewhat different to the Verilator VCD dump even if the behaviour is correct, because the memory controller will have greater variability in its latency.

zarubaf, 2019-10-09:
I assume this issue is resolved. In case it doesn't and you need further information please feel free to re-open or open a new one.

closed 2019-10-09 by zarubaf