openhwfoundation/cva6 #271: Error about generating a bitstream for FPGA

JunnanLi, 2019-07-01:
there is one error when I try to generate a bitstream based on the readme i.e.,
"source tcl/run.tcl
# set partNumber $::env(XILINX_PART)
# set boardName  $::env(XILINX_BOARD)
# set ipName xlnx_axi_clock_converter
# create_project $ipName . -force -part $partNumber
# set_property board_part $boardName [current_project]
ERROR: [Board 49-71] **The board_part definition was not found for digilentinc.com:genesys2:part0:1.1.** The project's board_part property was not set, but the project's part property was set to xc7k325tffg900-2. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store."

zarubaf, 2019-07-01:
You seem to miss the board files. See here how to install them: https://reference.digilentinc.com/vivado/installing-vivado/start (Step 3 in particular). That should resolve your problems.

zarubaf, 2019-07-10:
I assume this is resolved since you got the debugger to work.

closed 2019-07-10 by zarubaf