openhwfoundation/cva6 #273: WFI

suppamax, 2019-07-04:
Hi,

I have a problem using the WFI instruction with Ariane 4.2.

My asm code looks like this
```
...
 200096a:	10500073          	wfi
 200096e:	0001                	nop
 2000970:	0001                	nop
 2000972:	0001                	nop
 2000974:	0001                	nop
 2000976:	0001                	nop
 2000978:	c09d                	beqz	s1,200099e <test+0xd4>
 200097a:	00144783          	lbu	a5,1(s0)
 ...
```

and this is the corresponding waveform
[image]

The 7 instructions after the WFI command has reached the csr_regfile are executed (even if the interrupt has not yet arrived). This results in the LBU instruction being executed, even if I would have expected it to be run after the reception of the irq (it reads data written by the peripheral generating the interrupt).

Is this the way Ariane is expected to work?
I didn't see this behaviour with the Ariane core released last October.

An obvious workaround for me is to add 7 NOPs after WFI.

zarubaf, 2019-07-04:
1. You can’t rely on `wfi` to actually wait for an interrupt. This instruction is a hint which could be legal treated as a `nop`.
2. I can’t remember changing anything in it though. Are you sure you don’t have an interrupt pending (even if masked off)? The code which clears the interrupt is this: https://github.com/pulp-platform/ariane/blob/e873743b547fbf39bd08f25a0bfa7db38eb61794/src/csr_regfile.sv#L879

suppamax, 2019-07-05:
Is there any other mechanism to put the core to sleep?

axel-h, 2019-07-05:
I'd expect that WFI is either a NOP or some kind of halt. But in this case no instruction after the WFI should be executed before the WFI actually hits. And it seem this is happening here, the core executes instruction afterwards before it stops. This is a strange behavior actually. Should there be a pipeline flush after a WFI or an explicit fence instruction that prevents further fetching - if this exists on RISC-V. Otherwise the WFI implementation should implement this behavior.

zarubaf, 2019-07-05:
@axel-h I don't think that RISC-V provides the semantic for what you want to achieve (at least I don't remember that the specification says something about it). Imho you would need a `fence` to force a pipeline flush. 
@suppamax That would be something you need to implement on the platform level (possibly exposed through a platform peripheral), so far there is no standardized approach governed by RISC-V and we do not provide anything alike.  Did you check why the core doesn't actually halt? Btw, the pipeline will still run and fill up, the `wfi` just stops the commit stage from retiring instructions. So most likeley your `lbu` will read stale data.

zarubaf, 2019-07-05:
@axel-h There is certainly no requirement that a `wfi` enforces consistency to the IO space.

suppamax, 2019-07-05:
Hi Florian,
the core eventually halts - not immediately as I expected but after 7 clock cycles.
As you mentioned, the memory access required by LBU is executed when the instruction reaches the EX stage, but the read data are written to the register only after the interrupt has arrived; and they're incorrect since they've been fetched too early.
If you say this is the way the core is expected to work, and given that the ISA specs don't enforce any particular behaviour, I guess I have to live with that.

But I would ask you to consider improving this in a future version of Ariane.

closed 2019-07-25 by zarubaf