openhwfoundation/cva6 #274: Error in debugging on FPGA

JunnanLi, 2019-07-05:
When I start openocd with the fpga/ariane.cfg configuration file, I meet the following error: 
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
**embedded:startup.tcl:21: Error: Unknown target type riscv, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, cortex_m, cortex_a, cortex_r4, arm11, ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, or1k, quark_x10xx,  or quark_d20xx
in procedure 'script' 
at file "embedded:startup.tcl", line 60
in procedure 'target' called at file "fpga/ariane.cfg", line 27
in procedure 'ocd_bouncer' 
at file "embedded:startup.tcl", line 21**

zarubaf, 2019-07-05:
I am not sure if the RISC-V patches are already upstream. In doubt try building this one https://github.com/riscv/riscv-openocd

JunnanLi, 2019-07-06:
Thanks. Now I can connect to the FTDI chip, but these is no "Info : accepting 'gdb' connection on tcp/3333" : 
**Open On-Chip Debugger 0.10.0+dev-00619-g91faf1a57 (2019-07-05-22:29)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=64, misa=0x800000000014112d
Info : Listening on port 3333 for gdb connections
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections**

zarubaf, 2019-07-10:
I am confused, is this the output of your openocd? If so it literally says `Info : Listening on port 3333 for gdb connections`.

closed 2019-07-25 by zarubaf