openhwfoundation/cva6 #275: Errors in making verilate

JunnanLi, 2019-07-08:
Hi,
I meet following errors in making verilate, and I have checked out the newest version of Ariane and its submodules (git submodule update --init --recursive):
**%Error: /home/lijunnan/code/ariane/src/fpga-support/rtl/SyncSpRamBeNx64.sv:165: syntax error, unexpected assert
%Warning-ASSIGNDLY: /home/lijunnan/code/ariane/tb/common/SimDTM.sv:38: Unsupported: Ignoring delay on this assignment/primitive.
%Warning-ASSIGNDLY: Use "/* verilator lint_off ASSIGNDLY */" and lint_on around source to disable this message.
%Warning-ASSIGNDLY: /home/lijunnan/code/ariane/tb/common/SimDTM.sv:39: Unsupported: Ignoring delay on this assignment/primitive.**

zarubaf, 2019-07-08:
What `verilator` Version are you using? I remember seeing a similar error with an old version. Maybe try `> 4`.

JunnanLi, 2019-07-09:
@zarubaf 
Thinks, I have resolved above error by install a verilator-4.016. However, I meet a new error:
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/include -I/opt/riscv/include -std=c++11 -I../tb/dpi   -c -o SimDTM.o ../tb/dpi/SimDTM.cc
../tb/dpi/SimDTM.cc:4:10: fatal error: fesvr/dtm.h: No such file or directory
 #include <fesvr/dtm.h>

zarubaf, 2019-07-10:
I think you are missing `riscv-fesvr` Make sure the `dtm.h` is in `$RISCV/include/fesvr`

hyf6661669, 2019-07-11:
@zarubaf 
@JunnanLi 
I think the reason is that there is no "ariane/tb/riscv-isa-sim/fesvr". It seems that you don't upload this directory. When I was doing verilator simulation, I downloaded that from https://github.com/riscv/riscv-isa-sim and evevry thing is OK.

csunclechen, 2019-07-11:
I had the same problem about verilator version. thanks.

csunclechen, 2019-07-11:
> Thanks

> @zarubaf
> Thinks, I have resolved above error by install a verilator-4.016. However, I meet a new error:
> g++ -I. -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd [...]

how did you update verilator from 3.9 to 4?
I used ubuntu apt to install verilator. 
It shows that 3.916 is already the latest version.

csunclechen, 2019-07-11:
> how did you update verilator from 3.9 to 4?
> I used ubuntu apt to install verilator.
> It shows that 3.916 is already the latest version.

I have found that in verilator website.

closed 2019-07-14 by zarubaf