openhwfoundation/cva6 #279: Can't boot latter versions of Ariane

panosperi, 2019-07-12:
I have been able to successfully boot  Ariane 4.0.0 in Genesys 2 board using both bbl and opensbi firmware. However the next versions  (4.1 and 4.2) don't seem to work(its stucks before the uart shows linux kernel messages and gives me console). 
Has anyone faced a similar problem?

zarubaf, 2019-07-14:
That is strange, do you know the difference to our setup? I can still boot Linux normally with the latest version.

panosperi, 2019-07-15:
The version of Linux kernel i am using is 5.1.0-rc7. What is the difference of your setup?

zarubaf, 2019-07-25:
We don't use opensbi. Maybe that is related to the missing `coutneren` (they are missing now, which still should make a valid implementation) which have been fixed in the later versions. Apart from that imho the situation got more stable actually.

closed 2019-10-09 by zarubaf