openhwfoundation/cva6 #282: Question of running a RISC-V ELF using the Verilator model

JunnanLi, 2019-07-14:
Hello,
@zarubaf 
I meet a trouble running a RISC-V ELF using the Verilator model. When I run the command "work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf", it always waits for "listening on port 33523". Do I need to establish a TCP connection?

zarubaf, 2019-07-14:
No, it just says you could for debugging purposes but you don't have to.

hyf6661669, 2019-07-15:
@JunnanLi  Running the RTL simulation would be very slow, you might need wait for more than 600 seconds.

JunnanLi, 2019-07-15:
@hyf6661669 
Thanks, I get it. And do you know what's the meaning of mcycle, minstret, and compled after xxx cycles?

hyf6661669, 2019-07-15:
@JunnanLi  
Mabye you can take my simulation time as a reference.
I'm running the verilator simulation on a server with Xeon E5-2690 V2 @3.00GHz. It took me about 800s to finish the `HelloAriane``` simulation without generating the vcd file, and about 1600s with generating the vcd file (by using the ```-v``` argument).

You can find that the simulation would write many data on your disk, so I think a good SSD would be helpful:
[image]

As for the meaning of mcycle, minstret, you can find that in ```riscv-privileged```.

JunnanLi, 2019-07-17:
@hyf6661669 ,
Thank you very much.

closed 2019-07-25 by zarubaf

emremehmet34, 2020-05-03:
> Hello,
> @zarubaf
> I meet a trouble running a RISC-V ELF using the Verilator model. When I run the command "work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf", it always waits for "listening on port 33523". Do I need to establish a TCP connection?

Actually after work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf" process I get the 'Hello Ariane!' result but if I would like to generate the .vcd file from this elf file, what I should write as a command?

kabylkas, 2020-05-03:
@emremehmet34, as far as I remember, you would first have to build with `DEBUG=1` enabled:
```
make verilate DEBUG=1
```
Then, run the simulator with `--vcd`
```
work-ver/Variane_testharness --vcd=dump.vcd $RISCV/riscv64-unknown-elf/bin/pk hello.elf
```

emremehmet34, 2020-05-03:
> @emremehmet34, as far as I remember, you would first have to build with `DEBUG=1` enabled:
> 
> ```
> make verilate DEBUG=1
> ```
> 
> Then, run the simulator with `--vcd`
> 
> ```
> work-ver/Variane_testharness --vcd=dump.vcd $RISCV/riscv64-unknown-elf/bin/pk hello.elf
> ```
Firstly thanks for your feedback but  Unfortunately it gives error;
work-ver/Variane_testharness: unrecognized option '--vcd=dump.vcd'
Usage: work-ver/Variane_testharness [EMULATOR OPTION]... [VERILOG PLUSARG]... [HOST OPTION]... BINARY [TARGET OPTION]...

hyf6661669, 2020-05-04:
@emremehmet34 
Just use ```-v``` option like this:
```
work-ver/Variane_testharness -v=dump.vcd $RISCV/riscv64-unknown-elf/bin/pk hello.elf
```

emremehmet34, 2020-05-04:
> @emremehmet34
> Just use `-v` option like this:
> 
> ```
> work-ver/Variane_testharness -v=dump.vcd $RISCV/riscv64-unknown-elf/bin/pk hello.elf
> ```
work-ver/Variane_testharness -v=dump.vcd /opt/riscv/toolchain/riscv64-unknown-elf/bin/pk hello.elf
work-ver/Variane_testharness: invalid option -- 'v'

Actually the thing I am trying is , I would like to make some arithmetic operations on ariane core and then and want to see that operations on alu and bootrom on gtkwave simulator(Thats why I am trying to get vcd file).
How can I achieve this?

kabylkas, 2020-05-04:
@emremehmet34, I have just tried it, it is working fine for me. 

It might give you `unrecognized option` message for `--vcd` or `-v` if you didn't build with `DEBUG=1` (make there wasn't a typo). 

Also, if you use -v option, don't use = (`work-ver/Variane_testharness -v dump.vcd`)

kabylkas, 2020-05-04:
> work-ver/Variane_testharness -v=dump.vcd /opt/riscv/toolchain/riscv64-unknown-elf/bin/pk hello.elf
> work-ver/Variane_testharness: invalid option -- 'v'
>
> Actually the thing I am trying is , I would [...]

You may want to write c or assembly code for whatever arithmetic operations you want to see. Then compile the binary and put it instead of `hello.elf` that you used in your command. You can then investigate the waveforms in the dumped vcd once you get it working.

emremehmet34, 2020-05-07:
> You may want to write c or assembly code for whatever arithmetic operations you want to see. Then compile the binary and put it instead of `hello.elf` that you used in your command. You can then [...]

Thanks for your help and I get the waveform from elf file but I have an another question;

I make a c file as making an simple addition operation and get the elf file from it as
"riscv64-unknown-elf-gcc hello.c -o hello.elf"

then make verilate and make verilate DEBUG=1 commands 
after that I wrote "work-ver/Variane_testharness rv64um-v-divuw" command and get the "trace_hart_00.dasm" file then get the logfile.txt

But after all that instructions I couldn't see the add operations I wrote in the logfile
Is there any wrong command I wrote?

emremehmet34, 2020-06-07:
> You may want to write c or assembly code for whatever arithmetic operations you want to see. Then compile the binary and put it instead of `hello.elf` that you used in your command. You can then [...]

I want to make an implementation on Ariane core and I have a simple C code including addition, subtraction divison and multiplication.
Then I want to use Ariane core to make calculations for this algorithm and make observations on gtkwave. How can I achieve this?

yinhanquan, 2020-12-04:
> Hello,
> @zarubaf
> I meet a trouble running a RISC-V ELF using the Verilator model. When I run the command "work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf", it always waits for "listening on port 33523". Do I need to establish a TCP connection?

I got the same question,and i have waited for a long time, it always waits for "This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 40599
" how can i sovle this problem?

Hassan-Raza-164, 2022-10-18:
the same problem is happening with me