openhwfoundation/cva6 #285: File list for vcs

Fan-f1bu, 2019-07-17:
Hi,
I want to simulate ariane with vcs and I found a file list called Flist.ariane.
However, I found there are two files missing:
openpiton/bootrom/baremetal/bootrom.sv
openpiton/bootrom/linux/bootrom_linux.sv
Did I use the right file list? Or which file list should I use for vcs?
Thank you very much!

JunnanLi, 2019-07-20:
Hi,
@Fan-f1bu 
Have you built the vcs simulation environment?

Fan-f1bu, 2019-07-20:
Thanks for response. 

Yes I have vcs. I can run the design now and I am trying to preload binary but haven't succeeded so far.
Since I am pretty new for those things, are there any advices you can give me?

JunnanLi, 2019-07-20:
@Fan-f1bu 
I encountered some errors building up the vcs simulation environment. Can you send me a copy of your makefile? My emai address is lijunnan@nudt.edu.cn, thank you very much!

Fan-f1bu, 2019-07-20:
My code is on the computer at my office. So I can only give you until next Monday.
Actually, I imitated from the makefile from a RISC-V core celled ibex. You can try yourself.

closed 2019-07-25 by zarubaf