openhwfoundation/cva6 #286: MMU/TLB/SFENCE.VMA in Ariane 4.2

suppamax, 2019-07-18:
Dear PULP team,

While testing the microkernel on Ariane v4.2 we are experiencing some weird behaviour which seems to be connected with the MMU/TLB/SFENCE implementation.
It happens very often that the testsuite (which should be pretty stable, since it has been successfully used with other platforms) crashes in a way that it is not possible to easily determine the exception cause.
By analyzing the few useful information we’ve been able to collect (we’ve seen an INSTR_PAGE_FAULT in a handful of failing tests), we’ve made some experiments and discovered that changing the code from

write satp
sfence.vma

to

sfence.vma
sfence.vma
write satp
sfence.vma
sfence.vma

makes the tests much more stable.

We’ve also tried 

sfence.vma
write satp

as suggested in the riscv privilege specs v1.10 (page 58) but this caused even more issues than the original version.

Have you already encountered such situations? Maybe you already have some improvements in this direction which might be useful in our case?

Thanks and best regards,
Max

zarubaf, 2019-07-18:
We have definitely nothing which would “improve” this. By other platforms you mean RISC-V platforms? It is definitely possible that this a bug in the core but judging from the information you private it is impossible for me to pinpoint anything in particular.

axel-h, 2019-08-10:
Seems we can close this, as further investigation showed the actualy issue is #301

closed 2019-08-12 by zarubaf