openhwfoundation/cva6 #293: Is there a list of compatible linux image for the Ariane

mathmax12, 2019-07-30:
Hi,
I just start with the Ariane.
1. I wonder which Linux kernel should I chose to build the Linux image, by which I can boot up the system on the Ariane core?
2. Does it support the NVMe driver?

Thanks.

Moschn, 2019-07-31:
We use the mainline linux kernel. I am not sure what the current version in the ariane-sdk repository is, but previously we booted with 4.19/4.20. The kernel obviously needs to be built with a RISC-V cross-compilation toolchain and it needs a bootloader first (we are using bbl for the moment).

As for NVMe, I guess you would need a NVMe peripheral first (which probably also needs a PCI bus).

mathmax12, 2019-07-31:
Thanks very much for the reply.

closed 2019-08-05 by Moschn