openhwfoundation/cva6 #295: verilator:not found

mathmax12, 2019-08-05:
Hi,
I followed the readme file.
Installed the RISC-V tools.
Checked the update of Ariane by using(git submodule update --init --recursive)
When I try "make verilate" the system throw:
"/bin/sh: 1: verilator: not found
[Verilator] Building Model
....
...
bin/sh: 1: verilator: not found
Makefile:378: recipe for target 'verilate' failed
make: *** [verilate] Error 127
"

Can anybody help me with this?

Moschn, 2019-08-06:
Verilator is not included in the ariane-sdk. It only includes the toolchain and the linux image. You can get verilator from here: https://www.veripool.org/wiki/verilator.

Take a look at our verilator CI [script](ci/install-verilator.sh) for an example on how to install verilator.

closed 2019-08-06 by Moschn

mathmax12, 2019-08-06:
Thanks. it works after I installed the verilator.