openhwfoundation/cva6 #297: INSTR_PAGE_FAULT vs INSTR_ACCESS_FAULT

suppamax, 2019-08-06:
Hi,
let's assume the MMU generates an INSTR_ACCESS_FAULT exception.
Is it possible that this exception propagates to I$ (icache_areq_i) and then to instr_queue (icache_dreq_i.ex) and there is renamed to INSTR_PAGE_FAULT (line 220)?

If yes, why?

Thanks,
Max

axel-h, 2019-08-10:
Seem this also relates to #301

suppamax, 2019-08-10:
I don’t agree.
In 301 we have a wrong predicted address and an incomplete flush.

What is described here is independent from both.

zarubaf, 2019-08-12:
Yes, your observation is correct. Mainly because originally we didn't support `PMAs` so the only exception which could occur in the instruction front-end was an instruction page fault. To save memory in the instruction queue we just keep a single bit to indicate whether this instruction threw an exception or not. 

The implementation should not be tricky but its rather the verification of its correct implementation which prevented it so far. Would you require this feature?

Steinegger, 2020-01-23:
I stumbled upon the same issue when I tried to add an additional exception to the mmu and received a page fault, no matter what the original exception was.

So I tried your PR, however, it doesn't seem to fully work yet.
I wrote a small test file to jump to the virtual address 0xAAAAAAAA55555555, effectively violating the constraint in the `instr_interface` of the mmu:
https://github.com/pulp-platform/ariane/blob/11867888f337b6bc03f604fc6253cc3a3229a41e/src/mmu.sv#L210-L212
However, the jump the only results in an illegal_instruction, which doesn't make sense here, I think.

I further noticed that the `icache_areq_o.fetch_valid` signal is set for all the other cases, but not in this first case, which might have something to with this? It looks like the wt_icache system requires the `fetch_valid` bit to be set to perform the check on the `ex.valid` https://github.com/pulp-platform/ariane/blob/11867888f337b6bc03f604fc6253cc3a3229a41e/src/cache_subsystem/wt_icache.sv#L228-L233

Am I missing something here?

commit 11867888 (2019-10-04): frontend: Fix #297 remapping of instruction access faults
commit e0cca2ab (2020-07-14): frontend: Fix #297 remapping of instruction access faults
closed 2020-07-14 by zarubaf