openhwfoundation/cva6 #298: Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf

mathmax12, 2019-08-06:
1. I installed the risc-v toochain, verilator.
2. I build the verilator model by, $ make verilate DEBUG
3. Then I can see the work-ver/Variane_testharness was produced.
4. Then I compiled the hello.c with "riscv64-unknown-elf-gcc hello.c -o hello.elf"
5. After that, a hello.elf was produced.
6. Then a "hang" happened after "work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf" 
"ariane/work-ver$./Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 37725

"
In my understanding, the  Verilator model, i.e. **Variane_testharness** is a C++ model of the core including a SystemVerilog wrapper.  There is no Linux image and running user application just like in a bare metal model. Is my understanding correct?

What may cause this hang (step 6)? do I need a JTAG using the  Verilator model?
 
Thanks

Moschn, 2019-08-07:
Variane_testharness is indeed a c++ model of the core. However, this is a full RTL simulation and  pretty slow (see #282).  

> I'm running the verilator simulation on a server with Xeon E5-2690 V2 @3.00GHz. It took me about 800s to finish the `HelloAriane` simulation without generating the vcd file, and about 1600s with generating the vcd file (by using the `-v` argument).

You are probably not observing a "hang". Just give it some time.

mathmax12, 2019-08-07:
I see.
It did work after a while.

Thanks.

closed 2019-08-08 by Moschn