openhwfoundation/cva6 #299: boot from flash

mathmax12, 2019-08-07:
Hi,
 I want to know is it possible to put the FPGA bitstream, zero boot loader, first boot loader, Linux image together to the flash?
is there such flow I can follow?

Thanks.

Moschn, 2019-08-08:
The bitstream includes the zero-stage boot loader in a ROM and as far as I know we had that working at some point. There should be tutorials by Xilinx to put a bitstream into flash.

We have not tried to put the first-stage bootloader (bbl) and the linux image in flash. It should certainly be possible if there is enough space. However, it would need some changes in the zero-stage bootloader such as adding a bare-metal driver for the SPI flash.

closed 2019-08-09 by Moschn

mathmax12, 2019-08-28:
@Moschn,

Can you show more details about this?
Thanks

Moschn, 2019-08-29:
There is a digilent tutorial for example: https://reference.digilentinc.com/learn/programmable-logic/tutorials/genesys-2-programming-guide/start

mathmax12, 2019-08-29:
Hi @Moschn ,
Thanks for.
I know the flow of programming the Genesys2 board through Quad SPI Flash.
My question is how to put the **bitstream, bootloader, and Linux image** together to generate a single file (e.g., .bin or .mcs) file, after which I can put this single file to the Quad SPI Flash for booting the Linux system.

I know the VIvado SDK can do that targeting the Zynq SOC (ARM hardcore).
Is there a way that we can do the same thing for Ariane core?

Moschn, 2019-08-30:
At the moment this is not possible. The zero stage bootloader explicitly looks for an SD card connected over SPI and loads the bbl+linux from there. 

In order to support loading bbl+linux from SPI flash one needs to add the following:
- write a driver in the zero stage bootloader for the SPI flash
- extract the bbl+linux image somehow from the flash
  - probably requires filesystem support

mathmax12, 2019-09-09:
Is there any documentation to follow?
Does the Ariane have any plan to do this?

Moschn, 2019-09-10:
We do not plan to add this in the near future AFAIK. 

If you want to get your hands dirty: [fpga/src/bootrom](fpga/src/bootrom).

The SPI driver should be really easy since SPI flash is directly addressable. You would need to connect it to the system bus first though. After that you need to read up on how xilinx manages their bitstream in the SPI flash and then find a way to put the boot image next to the bitstream (thats where i think we would need a file system).