openhwfoundation/cva6 #301: Architecturally visible exception in mispredicted branch

niwis, 2019-08-09:
Hi,

We observe a `ret` instruction where the return address is wrongly predicted as `0x0` by the return address stack. The speculative execution at `0x0` triggers an `INSTRUCTION_PAGE_FAULT`, since `0x0` is not mapped. However, since it is mispredicted, it should be flushed once the misprection is detected. Yet, we see the exception occur nonetheless.

We've included a buffer that tracks some internal signals of the frontend:
<details><summary>Frontend Buffer</summary>

| Time | icache\_dreq\_o\.vaddr | priv\_lvl | Instr\[0\] | Instr\[1\] | Active Signals                                                                                                               |
|------|------------------------|-----------|------------|------------|-----------------------------------------------------------------------------------------------------------------------|
| 0    | 0xffffffff810a0000     | S         | 0x0        | 0x0        | \-                                                                                                                    |
| \-1  | 0x2c430                | U         | 0x0        | 0x0        | \-                                                                                                                    |
| \-2  | 0x2c42c                | U         | 0x87aad3df | 0x0        | \-                                                                                                                    |
| \-3  | 0x2c42a                | U         | 0xd3dfe0ef | 0x87aa     | if\_ready, icache\_dreq\_o\.kill\_s2, bp\_valid, instr\_valid\[0\], instr\_valid\[1\], rvi\_call\[0\], rvi\_jump\[0\] |
| \-4  | 0x2d6f4                | U         | 0xe0ef4501 | 0x0        | if\_ready, instr\_valid\[0\]                                                                                          |
| \-5  | 0x2d6f0                | U         | 0x45814601 | 0x4581     | if\_ready, instr\_valid\[0\], instr\_valid\[1\]                                                                       |
| \-6  | 0x2d6ec                | U         | 0xfec42783 | 0x86be     | if\_ready, instr\_valid\[0\], instr\_valid\[1\]                                                                       |
| \-7  | 0x2d6e8                | U         | 0x2783     | 0x0        | if\_ready                                                                                                             |
| \-8  | 0x2d6e4                | U         | 0x71398082 | 0x7139     | if\_ready, rvc\_return\[0\]                                                                                           |
| \-9  | 0x2d6e2                | U         | 0x71398082 | 0x7139     | if\_ready, rvc\_return\[0\]                                                                                           |
| \-10 | 0x4                    | U         | 0x71398082 | 0x7139     | icache\_dreq\_o\.kill\_s1, icache\_dreq\_o\.kill\_s2, is\_mispredict, flush\_i, rvc\_return\[0\]                      |
| \-11 | 0x0                    | U         | 0x71398082 | 0x7139     | icache\_dreq\_o\.kill\_s2, bp\_valid, instr\_valid\[0\], instr\_valid\[1\], rvc\_return\[0\]                          |
| \-12 | 0x2d764                | U         | 0x61217442 | 0x6121     | instr\_valid\[0\], instr\_valid\[1\]                                                                                  |
| \-13 | 0x2d760                | U         | 0x61217442 | 0x6121     | if\_ready                                                                                                             |
| \-14 | 0x2d75c                | U         | 0x61217442 | 0x6121     | \-                                                                                                                    |
| \-15 | 0x2d764                | U         | 0x61217442 | 0x6121     | icache\_dreq\_o\.kill\_s1, icache\_dreq\_o\.kill\_s2, replay, instr\_valid\[0\], instr\_valid\[1\]                    |
| \-16 | 0x2d760                | U         | 0x70e2853e | 0x70e2     | if\_ready, instr\_valid\[0\], instr\_valid\[1\]                                                                       |
| \-17 | 0x2d75c                | U         | 0xf8e7d8e3 | 0x4781     | if\_ready, instr\_valid\[0\], instr\_valid\[1\], rvi\_branch\[0\]                                                     |
| \-18 | 0x2d758                | U         | 0x07800793 | 0x0        | if\_ready, instr\_valid\[0\]                                                                                          |
| \-19 | 0x2d754                | U         | 0x0007871b | 0x0        | if\_ready, instr\_valid\[0\]                                                                                          |
| \-20 | 0x2d750                | U         | 0xfec42783 | 0x0        | if\_ready, instr\_valid\[0\]                                                                                          |
| \-21 | 0x2d74c                | U         | 0xfec42783 | 0x0        | if\_ready, instr\_valid\[0\]                                                                                          |
| \-22 | 0x2d748                | U         | 0x26232785 | 0x0        | if\_ready, instr\_valid\[0\]                                                                                          |
| \-23 | 0x2d744                | U         | 0xfec42783 | 0x0        | if\_ready, instr\_valid\[0\]                                                                                          |

</details>

The contents get updated with each new `fetch_address` while we are in or come from u-mode. It can be seen that `0x0` is predicted (at `Time = -11`) but discarded (`is_mispredict`) in the following step. The `0x0` instruction fetch makes it's way through the I$ to the PTW, where the invalid PTE is found, the exception created (`ptw.sv:222`) and eventually - unexpectedly - retired. We would have expected that it got flushed before retirement.

With best regards,
Nils

zarubaf, 2019-08-12:
Thanks for the report. Can you tell me on which commit of Ariane you are working? 

Definitely, exception which has been mispredicted should be flushed away once the mispredict has been detected. At time `-10` it seems that the `flush_i` signal correctly goes high. The question would be where the state is not correctly flushed out. I assume that `0xffffffff810a0000` is the S-Mode trap handler, right?

Could you produce a microprogram which triggers this behavior? I unfortunately don't see another way to debug this except for having full visibility in RTL simulation.

niwis, 2019-08-12:
> Can you tell me on which commit of Ariane you are working?

That's v4.2.0 

>  I assume that `0xffffffff810a0000` is the S-Mode trap handler, right?

Yes, that is correct

> Could you produce a microprogram which triggers this behavior? I unfortunately don't see another way to debug this except for having full visibility in RTL simulation.

I'll definitely try that. The problem is, that the exception does not seem to occur completely deterministically. This seems to be a very specific scenario and we are not completely aware of all the necessary conditions to reproduce this yet.

niwis, 2019-08-15:
So while we are still working a minimal test case to reproduce this issue, some further information was revealed.

While we know that the exception we encounter originates from `ptw.sv:222` when attempting to access the mispredicted zero-page, we actually see the exception retiring together with the next instruction which triggers a PTW (`0x2c42a` at time `-3`). This false association seems to occur within the MMU: `ptw_error` is raised (because of the mispredicted zero-page) while `icache_areq_i.fetch_vaddr = 0x2c42a`.

Can you think of a scenario where PTW is still busy walking an old, already killed iaddress when the I$ already requests a new, valid translation? (Our page table is stored in external DRAM, so PTWs might take some time)

zarubaf, 2020-01-22:
@niwis Do you have any update on this?

niwis, 2020-01-22:
No. Back then I implemented a work around and now on our Genesys2 platform the issue doesn't appear. Specific timing of memory accesses seems critical for this to happen

suppamax, 2020-03-18:
Hi both,

following @niwis line of thought...
Let's assume PTW is trying to convert address 0x0 while it receives a flush.

```
        // -------
        // Flush
        // -------
        // should we have flushed before we got an rvalid, wait for it until going back to IDLE
        if (flush_i) begin
            // on a flush check whether we are
            // 1. in the PTE Lookup check whether we still need to wait for an rvalid
            // 2. waiting for a grant, if so: wait for it
            // if not, go back to idle
            if ((state_q == PTE_LOOKUP && !data_rvalid_q) || ((state_q == WAIT_GRANT) && req_port_i.data_gnt))
                state_d = WAIT_RVALID;
            else
                state_d = IDLE;
        end
```
If state_d == WAIT_GRANT and req_port_i.data_gnt == 0, state will remain in WAIT_GRANT and the flush will substantially not have effect here.
Or am I missing some bits?

suppamax, 2020-03-24:
Hi,

following @niwis reasoning...
Wouldn't it be possible that
- PTW starts walking to translate address 0x0
- the request gets flushed (PTW back to IDLE)
- cache_ctrl keeps running (req_port_i.kill_req, driven by PTW, is always 0 and flush is 0 since it is set only upon reception of fence commands)
- the subsequent PTW request will get the old response from the cache 

What do you think?

masgia, 2020-07-09:
Hi,

I was investigating this issue and, as already mentioned, it seems like a PTW error triggered by a wrong prediction could propagate to the following address conversion.

I've found this piece of mmu code a bit suspicious
```
            // watch out for exceptions happening during walking the page table
            if (ptw_active && walking_instr) begin
                icache_areq_o.fetch_valid = ptw_error;
                icache_areq_o.fetch_exception = {riscv::INSTR_PAGE_FAULT, {{64-riscv::VLEN{1'b0}}, update_vaddr}, 1'b1};
            end
```
I don't understand why icache_areq_o.fetch_exception should contain a valid exception by default.
I tried a very simple modification
```
-                icache_areq_o.fetch_exception = {riscv::INSTR_PAGE_FAULT, {25'b0, update_vaddr}, 1'b1};
+               icache_areq_o.fetch_exception = {riscv::INSTR_PAGE_FAULT, {25'b0, update_vaddr}, ptw_error};
```
and the INSTR_PAGE_FAULT error disappeared.

So my questions:
- what's the reason for the original implementation?
- do you see any drawback in my fix (which requires anyway more investigations)?

MikeOpenHWGroup, 2023-02-18:
Hi @suppamax, it looks like this issue was resolved, can you confirm?

closed 2023-06-13 by cathales