openhwfoundation/cva6 #304: When there is a lsu instruction, ptw cannot handle itlb_miss

hyf6661669, 2019-08-13:
In src/ptw.sv:
```
// if we got an ITLB miss
                if (enable_translation_i & itlb_access_i & ~itlb_hit_i & ~dtlb_access_i) begin
                    ptw_pptr_n          = {satp_ppn_i, itlb_vaddr_i[38:30], 3'b0};
                    is_instr_ptw_n      = 1'b1;
                    tlb_update_asid_n   = asid_i;
                    vaddr_n             = itlb_vaddr_i;
                    state_d             = WAIT_GRANT;
                    itlb_miss_o         = 1'b1;
                // we got an DTLB miss
                end else if (en_ld_st_translation_i & dtlb_access_i & ~dtlb_hit_i) begin
                    ptw_pptr_n          = {satp_ppn_i, dtlb_vaddr_i[38:30], 3'b0};
                    tlb_update_asid_n   = asid_i;
                    vaddr_n             = dtlb_vaddr_i;
                    state_d             = WAIT_GRANT;
                    dtlb_miss_o         = 1'b1;
                end
```
When there is an lsu instruction, but the lsu instruction doesn't meet dtlb_miss, then we get 
dtlb_access_i = dtlb_hit_i = 1. If we get a itlb_miss at the same time, then we cannot handle this. But actually the ptw is not working.

zarubaf, 2019-08-19:
You are right that this is priority encoded, but the convention for the PTW interface is that the TLB will keep the miss request high as long as there is an unresolved TLB miss. Hence, the worst that could happen is for the PTW walker to serve an instruction TLB miss and afterwards the data TLB miss.

hyf6661669, 2019-08-20:
@zarubaf 
Does this mean that lsu instructions have higher priority than ptw, when they both need to access the dcache?

closed 2019-09-01 by hyf6661669