openhwfoundation/cva6 #305: Elaborating RTL using Cadence Jaspergold

hyunsuchae, 2019-08-14:
Hi,
I am trying the elaborate the design using Cadence JasperGold.
I run into an error saying "expression has 6216 elements; expected 15" in the src/ariane.sv file line 223: 
[image]

I think this problem has to do with ariane_pkg.sv, where some of the "ariane_cfg_t" elements are in arrays, but I can't exactly figure out how to fix this.

zarubaf, 2019-08-19:
Yes indeed it seems that the tool has problems understanding the width of the vectors inside the struct. I unfortunately don't have experience using Cadence tools for either synthesis or formal verification. It would be interesting for me to follow up on this as I think supporting Cadence is important as well. Maybe somebody else has more experience with that?

hyunsuchae, 2019-08-20:
What I think is happening is..

The error message I get is:
[image]

which is saying that there are **6216** elements but there are supposed to be **15** elements.

The part that has the error in "src/ariane.sv" file is the line 223 that looks like this:
[image]

In this top module "src/ariane.sv", ArianeCfg is defined in the line 18 like this:
[image]

Sine the ariane_pkg is imported and used, I looked at the file that defines this, which is "ariane_pkg.sv":
[image]

What I have found is that when the number of the struct packed elements in "araine_cfg_t" is 15 in total (from line 37~54 in the picture above, "ariane_pkg.sv").
But when you count all the bits of these elements, there are 6216 bits in total. 
For example, the line 42, since _NrMaxRules_ is 16, and it's 2d array, it thinks that it is 16*64 = 1024 bits. And if I do this for all 15 elements, it make total bits of 6216.

So, I understand where this error saying "there are 6216 elements but expected 15" is coming from. But I have no idea why, and how to fix this issue.

closed 2020-01-22 by zarubaf

RakeshpatilLB, 2025-06-27:
Hi @hyunsuchae , i am trying to elaborate the cva6 design (core only ) on the JASPER GOLD tool. The tool is able to read the files succesfully, but the elaboration has produced a lot of bit width mismatch warnings ,similar to what you encountered.  I hope you might have figured out the solution for your issue, if yes could you please help me resolving my issue...?

Warnings file:

[jasper (1).log](https://github.com/user-attachments/files/20953639/jasper.1.log)