openhwfoundation/cva6 #309: Linux boot get stuck

mathmax12, 2019-08-21:
Hi,
The pre-built linux image Ariane 4.2 on the Genesys2 board.
Then I followed the Ariane-SDK (https://github.com/pulp-platform/ariane-sdk/issues?q=is%3Aissue+is%3Aclosed) generate bootloarder and linux image.

Each time the system booting will get stuck at "mmcblk0: mmc0:0000 SC16G 14.8 GiB" (Below is the booting information).
Is there any differece betweeen Ariane 4.2  and Ariane-SDK?

/**********
Hello World!
init SPI
status: 0x0000000000000025
status: 0x0000000000000025
SPI initialized!
initializing SD...
SD command cmd0         response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 00
sd initialized!
gpt partition table header:
        signature:      5452415020494645
        revision:       00010000
        size:           0000005C
        crc_header:     86AE3791
        reserved:       00000000
        current lba:    0000000000000001
        backup lda:     0000000001DACBFF
        partition entries lba:          0000000000000002
        number partition entries:       00000080
        size partition entries:         00000080
gpt partition entry 00
[... 95 lines cut ...]
[   24.082790] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[   24.157852] 10000000.uart: ttyS0 at MMIO 0x10000000 (irq = 1, base_baud = 3125000) is a TI16750
[   25.010031] printk: console [ttyS0] enabled
[   25.079376] xilinx_spi 20000000.xps-spi: at 0x20000000 mapped to 0x(____ptrval____), irq=2
[   25.119963] libphy: Fixed MDIO Bus: probed
[   25.148737] lowrisc-digilent-ethernet: Lowrisc ethernet platform (30000000-30007FFF) mapped to ffffffd004028000
[   25.203203] libphy: GPIO Bitbanged LowRISC: probed
[   25.214430] Probing lowrisc-0:01 (address 1)
[   25.229397] RTL8211E Gigabit Ethernet lowrisc-0:01: attached PHY driver [RTL8211E Gigabit Ethernet] (mii_bus:phy_addr=lowrisc-0:01, irq=POLL)
[   25.281512] lowrisc-eth 30000000.lowrisc-eth: Lowrisc Ether100MHz registered
[   25.303639] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[   25.374787] mmc_spi spi0.0: SD/MMC host mmc0, no WP, no poweroff, cd polling
[   25.404193] usbcore: registered new interface driver usbhid
[   25.416370] usbhid: USB HID core driver
[   25.478670] NET: Registered protocol family 10
[   25.562892] Segment Routing with IPv6
[   25.575240] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   25.622110] NET: Registered protocol family 17
[   25.659119] Key type dns_resolver registered
[   26.187830] Freeing unused kernel memory: 6088K
[   26.197912] This architecture does not have kernel memory protection.
[   26.211728] Run /init as init process
[   26.278820] mmc0: host does not support reading read-only switch, assuming write-enable
[   26.296543] mmc0: new SDHC card on SPI
[   26.380244] mmcblk0: mmc0:0000 SC16G 14.8 GiB

zarubaf, 2019-09-13:
Are you still facing that issue? My initial guess would be that the spi block device driver is somehow broken. Does it consistently not boot?

mathmax12, 2019-09-13:
Yes.
If the FPGA bitstream keeps the same, using the pre-build Linux image Ariane 4.2  will work. 
My own build with the Ariane-SDK doesn't work.

This problem persists. Here are some details that I found https://github.com/pulp-platform/ariane-sdk/issues/23#issuecomment-525485651
But still wouldn't work.

closed 2020-01-22 by zarubaf