openhwfoundation/cva6 #317: Using GDB to debug `rv64ui-v-add` on FPGA

jmason827, 2019-09-09:
I've loaded a bitstream for Ariane onto my FPGA. It seems to be working fine -- I'm able to boot Linux using an SD card.

I want to run some of the `riscv-tests`. In particular, I've compiled `rv64ui-v-add`. At first, everything seems OK; we start at PC = 0x8000_0000:

```
(gdb) load
Loading section .text.init, size 0x174 lma 0x80000000
Loading section .tohost, size 0x48 lma 0x80001000
Loading section .text, size 0xfac lma 0x80002000
Loading section .rodata.str1.8, size 0x1e0 lma 0x80002fb0
Start address 0x80000000, load size 4936
Transfer rate: 3 KB/sec, 1234 bytes/write.
(gdb) where
#0  _start ()
	at /home/jeff/riscv/riscv-tools/riscv-tests/isa/../env/v/entry.S:19
(gdb) x/i 0x80000000
=> 0x80000000 <_start>:    j    0x8000000c <handle_reset>
```

This squares with `rv64ui-v-add.dump`:

```
Disassembly of section .text.init:

0000000080000000 <_start>:
	80000000:   00c0006f            	j   	8000000c <handle_reset>

0000000080000004 <nmi_vector>:
	80000004:   2600206f            	j   	80002264 <wtf>

0000000080000008 <trap_vector>:
	80000008:   25c0206f            	j   	80002264 <wtf>

000000008000000c <handle_reset>:
	8000000c:   00000297            	auipc   t0,0x0
	80000010:   ffc28293            	addi	t0,t0,-4 # 80000008 <trap_vector>
``` 

But then if I try to place a breakpoint at `handle_reset` and do `(gdb) continue`, GDB never regains control. No breakpoint is ever triggered. I've tried placing breakpoints elsewhere in the program, but the result is always the same.

If, on the other hand, I do `(gdb) step` from the very beginning, I get the following result, where it appears to regain control at a nonsensical PC value:

```
(gdb) where
#0  _start () at /home/jeff/riscv/riscv-tools/riscv-tests/isa/../env/v/entry.S:19
(gdb) x/i 0x80000000
=> 0x80000000 <_start>:    j    0x8000000c <handle_reset>
(gdb) step
0xffffffffffe000cc in ?? ()
(gdb) where
#0  0xffffffffffe000cc in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

I suspect I must be misunderstanding some aspect of the `*.S` file compilations. Why isn't GDB able to reliably step through these tests?

zarubaf, 2019-10-09:
Hm, that probably indicates a problem in the debug logic somewhere. At least stopping should always be possible. Can it be that you are accidentally fetching from an address in your system which is not mapped e.g. a load or store is never returned?

jmason827, 2019-10-14:
Thanks for your reply, Florian. I'm going to close this issue for now because the team is working with Ariane in ways that don't require OpenOCD and I'm not sure when/if we'll get back to it. I'll reopen if we do start again with OpenOCD.

closed 2019-10-14 by jmason827

RaphaelKlink, 2020-03-26:
It depends on the version of openocd.
Warning: This version of OpenOCD does not support address translation. To debug on virtual addresses, please update to the latest version.
But I do not think I ever saw a RISC-V OpenOCD Version, which does support it.