openhwfoundation/cva6 #319: Predicted address in case of empty RAS

suppamax, 2019-09-11:
Hi,
this scenario is not clear to me.
In case of a RET instruction when the RAS is empty, npc will be set to 0.
```
          // return
          4'b0100: begin
            // make sure to only alter the RAS if we actually consumed the instruction
            ras_pop = ras_predict.valid & instr_queue_consumed[i];
            ras_push = 1'b0;
            predict_address = ras_predict.ra;
            cf_type[i] = ariane_pkg::Return;
          end
...
    // or reduce struct
    always_comb begin
      bp_valid = 1'b0;
      for (int i = 0; i < INSTR_PER_FETCH; i++) bp_valid |= (cf_type[i] != NoCF);
    end
```
even if ras_predict.valid=0, cf_type=Return, which will result in having bp_valid=1, which will result in npc_d = predict_address (even if invalid).
Is this correct?

zarubaf, 2019-09-13:
Yes, you are right. That case could indeed be improved by gating the `bp_valid` signal to only be asserted when the Return is actually valid predicted.

MikeOpenHWGroup, 2023-02-18:
Hi @suppamax, it _looks_ like this issue was resolved, can you confirm?

closed 2023-02-18 by suppamax