openhwfoundation/cva6 #321: Exit from WFI

suppamax, 2019-09-19:
Hi,

in csr_regfile.sv I see this condition is used to exit from WFI
```        
if (|mip_q || debug_req_i || irq_i[1]) begin
            wfi_d = 1'b0;
```
I don't understand why the core should be waken up in case of a pending interrupt, even if it has not been enabled via MIE, or in case of an external irq for supervisor mode, again unmasked.
In case an interrupt is received, the code will continue its execution but might not enter the ISR.

zarubaf, 2019-09-27:
As far as I remember the spec once said that any pending interrupt (even when not enabled) should un-stall the core. In the newer version the specification is much clearer on the behavior. Still, you can't place any assumptions on the behavior of the `wfi` it could be implemented as a `nop`. Agreeing that this should be fixed in the next version.

MikeOpenHWGroup, 2023-02-18:
Hi @suppamax , this _looks_ like it was resolved, can you confirm?

closed 2023-02-18 by suppamax