openhwfoundation/cva6 #323: Frequency in ASIC?

hyf6661669, 2019-09-26:
Could you tell any information about the frequency that Ariane could get in ASIC implementation?
Much thanks.

hyf6661669, 2019-09-26:
I found your paper here:

https://ieeexplore.ieee.org/document/8777130
https://arxiv.org/pdf/1904.05442.pdf

Excellent work!

hyf6661669, 2019-09-27:
@zarubaf  
Could you tell me that 1.7GHz is reached with or without RV64FD ?

[image]

zarubaf, 2019-09-27:
Thanks. The reported 1.7 GHz were without F and D. The silicon measurements are from an earlier tape-out which didn't support the floating point extension. It will most probably be a bit slower with the FPU although the FPU can be flexibly pipelined to suit your area/timing/latency requirements.

closed 2019-09-27 by hyf6661669

polkeemfungible, 2019-10-03:
Hi, are there any examples of how to pipeline the FPU and find those registers to specify to synthesis scripts for register retiming?

Moschn, 2019-10-04:
The [fpu repository](https://github.com/pulp-platform/fpnew) contains documentation on how to pipeline the FPU:

https://github.com/pulp-platform/fpnew/blob/master/docs/README.md#implementation---implementation-options

https://github.com/pulp-platform/fpnew/blob/master/docs/README.md#pipeconfig---pipeline-register-placement

https://github.com/pulp-platform/fpnew/blob/master/docs/README.md#pipelining