openhwfoundation/cva6 #324: The asid source when sfence.vma flush TLB

TianyangL, 2019-09-27:
Hi, why does TLB flush need asid from CSR when executing sfence. Vma instruction? I think asid should come from sfence. vma  rs2.

TianyangL, 2019-09-27:
Sorry. I just mistakenly posted the same problem multiple times due to an Internet error. I've closed them all.

zarubaf, 2019-09-27:
Does it? Can you point me to the code?

TianyangL, 2019-09-29:
In file tlb.sv,as shown below,in line102amd line104, use lu_asid_i when flush TLB. I found that lu_asid_i comes from CSR in ex_stage(ariane.sv)->load_store_unit(ex_stage.sv)->mmu(load_store_unit.sv)->tlb(tlb.sv)
 92     // ------------------
 93     // Update and Flush
 94     // ------------------
 95     always_comb begin : update_flush
 96         tags_n    = tags_q;
 97         content_n = content_q;
 98 
 99         for (int unsigned i = 0; i < TLB_ENTRIES; i++) begin
100             if (flush_i) begin
101                 // invalidate logic
102                 if (lu_asid_i == 1'b0) // flush everything if ASID is 0
103                     tags_n[i].valid = 1'b0;
104                 else if (lu_asid_i == tags_q[i].asid) // just flush entries from this ASID
105                     tags_n[i].valid = 1'b0;
106 
107             // normal replacement
108             end else if (update_i.valid & replace_en[i]) begin

suppamax, 2019-12-20:
My impression is that the SFENCE implementation follows the privileged specs v1.9, not the v1.10.
But even in this case, looks like the SFENCE is not fully implemented, as there is no rs1 (vaddr) field handling.

zarubaf, 2020-01-22:
Let's track this at #364.

closed 2020-01-22 by zarubaf