openhwfoundation/cva6 #328: Confused about load_unit.sv

hyf6661669, 2019-09-30:
[image]

I am confused about the logic here.
If a load instruction meets an exception(ex_i.valid = 1), and there is a new load instruction in lsu_bypass(valid_i = 1), it seems that the value of pop_ld_o doesn't matter. Because the early load instruction that has an exception will flush the whole pipeline and make the cpu enter the exception handler. So we don't have to care about the later load instruction in lsu_bypass.

zarubaf, 2019-10-09:
Yes I guess you are right. If we flush the `lsu_bypass` module. Would be great if you could try this fix on an FPGA and see whether it still works.

closed 2019-10-10 by hyf6661669

zhaoxiahust, 2020-03-04:
Also solved my concern. The same thing happens in the Store unit. Thanks!