openhwfoundation/cva6 #331: Linux booting gets stuck

Rahul-Kande, 2019-10-02:
Hello, 

I am trying to boot linux on Ariane SoC, here are the steps I followed: 
1) Programmed the Genesys 2 (https://www.mouser.com/ProductDetail/Digilent/410-300?qs=%2Fha2pyFaduifbWBD9JYWEMg4vNtAQsT%2FdPF0KoRwdpMkk%252BLPx0%2Frng%3D%3D) board with the ariane_xilinx.msc file from Ariane 4.2 release (https://github.com/pulp-platform/ariane/releases)
2) Followed the 'Booting from SD card' steps from here (https://github.com/pulp-platform/ariane-sdk) and used the bbl.bin file from the Ariane 4.2 release (https://github.com/pulp-platform/ariane-sdk/releases)
3) Plugged in the SD card, connected a terminal to the USB serial device opened by the UART port on the FPGA and started the FPGA. 
The terminal outputs the following and **gets stuck after 'copying boot image ......... done!'**
`Hello World!
init SPI
status: 0x0000000000000025
status: 0x0000000000000025
SPI initialized!
initializing SD...
SD command cmd0         response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 00
sd initialized!
gpt partition table header:
        signature:      5452415020494645
        revision:       00010000
        size:           0000005C
        crc_header:     17BCEAE1
        reserved:       00000000
        current lba:    0000000000000001
        backup lda:     0000000003B723FF
        partition entries lba:          0000000000000002
        number partition entries:       00000080
        size partition entries:         00000080
gpt partition entry 00
        partition type guid:    D5F7127456A1134B81DC867174929325
        partition guid:         CA06C1C8AE42B04DBF62C5EFF2F9CE16
        first lba:      0000000000000800
        last lba:       00000000000107FF
        attributes:     0000000000000000
        name:   000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 01
        partition type guid:    AF3DC60F838472478E793D69D8477DE4
        partition guid:         891927C5ECD4C740B95993BE66B15B97
        first lba:      0000000000010800
        last lba:       0000000003B723DE
        attributes:     0000000000000000
        name:   000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 02
        partition type guid:    00000000000000000000000000000000
        partition guid:         00000000000000000000000000000000
        first lba:      0000000000000000
        last lba:       0000000000000000
        attributes:     0000000000000000
        name:   000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 03
        partition type guid:    00000000000000000000000000000000
        partition guid:         00000000000000000000000000000000
        first lba:      0000000000000000
        last lba:       0000000000000000
        attributes:     0000000000000000
        name:   000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
copying boot image ................................ done!
`

Can someone please help me resolving this issue or suggest alternate methods to get the linux booted up on the FPGA?
 
Thanking you in advance

Rahul-Kande, 2019-10-03:
Hello, 
Thank you for the quick response. 
I did the following with respect to the sd card: 
1)  sudo fdisk -l
This showed that the SD card is in /dev/sdc
2) sudo sgdisk --clear --new=1:2048:67583 --new=2 --typecode=1:3000 --typecode=2:8300 /dev/sdc
3) sudo dd if=bbl.bin of=/dev/sdc1 status=progress oflag=sync bs=1M
The bbl.bin file I used here is directly from the sdk release version 4.2

Did I miss any steps or did something wrong? Is there a way to verify that the SD card is loaded the right way?

Moschn, 2019-10-03:
You seem to be doing the correct steps. Your debug output also shows that the SD card can be read and it seems to be partitioned correctly. 

You could attach gdb with openocd and check what the core is doing. But the procedure is a bit involved.

Rahul-Kande, 2019-10-03:
Currently, I do not have the setup for Openocd. I will try the Openocd option and get back to you. 

But something I do not understand is, I do not see any place where things could have gone wrong, I am using the release version .msb file, .bin file, and the same FPGA and there is nothing like machine dependency or toolchain dependency which could have gone wrong. 

Anyways, thank you for your support regarding this issue. Is it ok if I keep this issue open?

Moschn, 2019-10-03:
Sure keep it open. 

Maybe try to use a different SD card? Apparently, some SD cards only support SPI mode up to a certain frequency. We never had any problems with our cards but if you have another SD card laying around, maybe try that one.

rahulkande, 2019-10-20:
The issue got resolved. It was because of bad partitioning of the SD card. I tried with a different SD card by properly partitioning it and it now works. 
Thank you for the support

closed 2019-10-21 by Moschn