openhwfoundation/cva6 #337: Error while generating the bitstream

rahulkande, 2019-10-20:
Hello, 

I am trying to generate the bitstream by myself for the Genesys-2 FPGA baord. But I am getting an error that the 'dm' is not declared. The following are the last few lines of the log : 
`
WARNING: [Vivado 12-818] No files matched '*/../src/common_cells/include/common_cells/registers.svh'
# set file_obj [get_files -of_objects [get_filesets sources_1] [list "*$file" "*/$registers"]]
# set_property -dict { file_type {Verilog Header} is_global_include 1} -objects $file_obj
# update_compile_order -fileset sources_1
# add_files -fileset constrs_1 -norecurse constraints/$project.xdc
# synth_design -rtl -name rtl_1
Command: synth_design -rtl -name rtl_1
Starting synth_design
Using part: xc7k325tffg900-2
WARNING: [Vivado_Tcl 4-393] The 'Synthesis' target of the following IPs are stale, please generate the output products using the generate_target or synth_ip command before running synth_design.
/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/fpga/xilinx/xlnx_mig_7_ddr3/ip/xlnx_mig_7_ddr3.xci

WARNING: [Vivado_Tcl 4-393] The 'Implementation' target of the following IPs are stale, please generate the output products using the generate_target or synth_ip command before running synth_design.
/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/fpga/xilinx/xlnx_mig_7_ddr3/ip/xlnx_mig_7_ddr3.xci

Top: ariane_xilinx
WARNING: [Synth 8-1921] elaboration system task fatal violates IEEE 1800 syntax [/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/src/common_cells/src/stream_arbiter_flushable.sv:77]
WARNING: [Synth 8-1858] constant expression cannot contain a hierarchical identifier [/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/src/fpu/src/fpnew_fma_multi.sv:61]
WARNING: [Synth 8-1858] constant expression cannot contain a hierarchical identifier [/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/src/fpu/src/fpnew_fma_multi.sv:62]
WARNING: [Synth 8-1858] constant expression cannot contain a hierarchical identifier [/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/src/fpu/src/fpnew_cast_multi.sv:67]
WARNING: [Synth 8-1858] constant expression cannot contain a hierarchical identifier [/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/src/fpu/src/fpnew_cast_multi.sv:68]
ERROR: [Synth 8-1031] dm is not declared [/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/include/ariane_pkg.sv:224]
ERROR: [Synth 8-1031] dm is not declared [/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/include/ariane_pkg.sv:229]
ERROR: [Synth 8-1031] dm is not declared [/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/include/ariane_pkg.sv:230]
ERROR: [Synth 8-2119] illegal context for assignment pattern [/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/include/ariane_pkg.sv:231]
INFO: [Synth 8-2350] module ariane_pkg ignored due to previous errors [/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/include/ariane_pkg.sv:824]
Failed to read verilog '/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/include/ariane_pkg.sv'
1 Infos, 7 Warnings, 0 Critical Warnings and 5 Errors encountered.
synth_design failed
ERROR: [Vivado_Tcl 4-5] Elaboration failed - please see the console for details
INFO: [Common 17-206] Exiting Vivado at Sat Oct 19 20:06:46 2019...
make[1]: *** [work-fpga/ariane_xilinx.bit] Error 1
make[1]: Leaving directory `/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_bitstream/temp/ariane/fpga'
make: *** [fpga] Error 2
`
I also got a lot of other errors when I tried to run the 'make fpga' command on the latest repo on master branch. 
For ex: 1) The line 27 in fpga/scripts/prologue.tcl was throwing a syntax error near '[' . I commented out this command from the tcl script since it is just something related to output prints
2) The line 91 in src/common _cells/src/lzc.sv was throwing syntax error near '. I looked at the latest version of this file in git and replaced that line with the latest copy to overcome the error. (The ariane soc was not getting the latest version of this file when i do git submodule and hence even a latest cloned version of ariane soc has this error)
Now I am getting this error that 'dm' not declared. 
There are also many warnings like stale IPs etc which overlooked for now. 

Since the master branch is having these issues, I wanted to try the release version 4.2 but that zipped release source code doesnt seem to have all the submodules. 
Can someone suggest me a way where I can easily generate the bitstream without having to resolve all these errors one by one ? 
PS: I know that there is a prebuilt bitstream in 4.2 release, but I want to create my own bitstream because I am planning on making some changes in the ariane core and generate bitstream for the changes. 

Thanking you in advance.

hyf6661669, 2019-10-21:
It seems that you lack the contents of ariane/src/riscv-dbg(https://github.com/pulp-platform/riscv-dbg/tree/master), which is used for debugging.

rahulkande, 2019-10-21:
I just checked the directory. I do have the debug repo

rahulkande, 2019-10-21:
If you are able to generate the bitsteam, will it be possible for you to send me a zipped version of your repo ? 
I tried the release version also, but it did not have these dependency repos and I dont know how to download all of them without the git submodule command

hyf6661669, 2019-10-21:
[image]

```dm``` is defined in ariane/src/riscv-dbg/src/dm_pkg.sv. Vivado said ```dm``` is not declared.

hyf6661669, 2019-10-21:
Sorry, I just use the pre-built bistream for tests. Maybe you need to wait the maintainers to answer.

rahulkande, 2019-10-21:
> [image]
> 
> `dm` is defined in ariane/src/riscv-dbg/src/dm_pkg.sv. Vivado said `dm` is not declared.

Yes, I know. But I dont why it is throwing that error :/

Moschn, 2019-10-21:
So you want to generate the bitstream, but you do not have access to the git command? This seems pretty esoteric since on almost all Unix (and windows) systems, git can be easily installed. 

How did you get all the submodules to get the trace? Did you just download the master of all submodules? Sadly, Ariane does not use all master branches but specific commits. All submodules and the required git commit can be found in the  `.gitmodules` file. So if you really want to do that, you could go on every single repository and download the exact version that is required.

Rahul-Kande, 2019-10-21:
> So you want to generate the bitstream, but you do not have access to the git command? This seems pretty esoteric since on almost all Unix (and windows) systems, git can be easily installed.
>
> How did [...]

When I said I could not use the 'git submodule' command, I meant that I could not use it on the release version (4.2) of the ariane. 
I was ofcourse able to do 'git submodule' if I directly cloned the master branch. But the issue is, I am getting all the errors and warnings from the master branch, so I thought maybe the release version might be stable.

Moschn, 2019-10-21:
Hmm I just recently built the entire image on my machine without issues. What is the output of `git status`?

Can you run `git submodule update --init --recursive` one more time?

Rahul-Kande, 2019-10-21:
Can you check your repo for these two issues : 
1) The line 27 in fpga/scripts/prologue.tcl is throwing a syntax error near '[' . I commented out this command from the tcl script since it is just something related to output prints
Does your repo have line 27 as : `set_msg_config -id {[Synth 8-4480]} -limit 1000` ? 
2) The line 91 in src/common _cells/src/lzc.sv is throwing syntax error near '. I looked at the latest version of this file in git and replaced that line with the latest copy to overcome the error. (The ariane soc was not getting the latest version of this file when i do git submodule and hence even a latest cloned version of ariane soc has this error)
Does your repo has line 91 as `assign cnt_o   = NUM_LEVELS > unsigned'(0) ? index_nodes[0] : $clog2(WIDTH)'(0);` ? 

I want to know if you have the same files and yet not getting errors or if you have different files. If it is the later, will it be possible for you to make a release or just send me the complete repo with all the dependency files? I dont care about any future updates, I immediately want a working repo in which I can generate bitstream myself

zarubaf, 2019-10-21:
We'll investigate these errors on our end. It might be the case that we broke something on the newer `master` as we added two new FPGA boards. Thanks for pointing out a potential problem.

What I don't understand is the weird request of somebody sending you the repo. Please, just check-out the `v4.2` tag and do a `git submodule update --init --recursive`.

zarubaf, 2019-10-21:
If in doubt do a fresh clone.

Rahul-Kande, 2019-10-21:
> We'll investigate these errors on our end. It might be the case that we broke something on the newer `master` as we added two new FPGA boards. Thanks for pointing out a potential problem.
>
> What I [...]

Ok, I am trying out the v4.2 commit. I will let you know the result.

Rahul-Kande, 2019-10-21:
I just did : 
 git clone -n https://github.com/pulp-platform/ariane.git
cd ariane
 git checkout e873743b547fbf39bd08f25a0bfa7db38eb61794    (v4.2 commit hash)
git submodule update --init --recursive
make fpga

and i am getting the following error : 
source scripts/prologue.tcl
# set project ariane
# create_project $project . -force -part $::env(XILINX_PART)
# set_property board_part $::env(XILINX_BOARD) [current_project]
# set_param general.maxThreads 8
# set_msg_config -id {[Synth 8-5858]} -new_severity "info"
# set_msg_config -id {[Synth 8-4480]} -limit 1000
ERROR: [Common 17-234] Invalid character '[' in message id specified '[Synth 8-4480]'.
ERROR: [Common 17-39] 'set_msg_config' failed due to earlier errors.

    while executing
"set_msg_config -id {[Synth 8-4480]} -limit 1000"
    (file "scripts/prologue.tcl" line 27)
INFO: [Common 17-206] Exiting Vivado at Mon Oct 21 07:30:43 2019...
make[1]: *** [work-fpga/ariane_xilinx.bit] Error 1
make[1]: Leaving directory `/home/grads/r/rahulkande/thesis/Ariane/git_beta/ariane_4_2/ariane/fpga'
make: *** [fpga] Error 2

Now, my question is, how are you guys able to generate the bitstream when I am getting this error?
Either my software has issues or we not using the same files. Which one is it?

zarubaf, 2019-10-21:
I am pretty confident that `v4.2` worked. It was used to generate the bitstreams you can download. What's your vivado version?

rahulkande, 2019-10-21:
> I am pretty confident that `v4.2` worked. It was used to generate the bitstreams you can download. What's your vivado version?

That could happen. My vivado seems to be pretty old (2015.2). Do you think this is the issue ? 
[rk@labpc ~/thesis/Ariane/git_beta/ariane_spike]$ vivado -version
Vivado v2015.2 (64-bit)
SW Build 1266856 on Fri Jun 26 16:35:25 MDT 2015
IP Build 1264090 on Wed Jun 24 14:22:01 MDT 2015
Copyright 1986-2015 Xilinx, Inc. All Rights Reserved.

zarubaf, 2019-10-21:
Yes, it seems so. The oldest version I’ve tested ist `2018.x`. I would suggest upgrading an retrying.

zarubaf, 2019-10-25:
Any progress?

rahulkande, 2019-10-29:
Thank you for the follow-up. I am in the process of acquiring license for the 2018 version. I will update once I run the make script with Vivado 2018.

closed 2020-01-22 by zarubaf