openhwfoundation/cva6 #338: Linux boot stuck on core without ethernet on Virtex Ultrascale

RaphaelKlink, 2019-10-22:
Hey,

i am trying to port the whole project to the VCU108 Evaluation Board.
I got most things working, but my boot gets stuck at the ethernet driver.
Is there an option to build the vmlinux without the ethernet driver?

Thank you for your help

Here is the last part of the UART output:

[    0.674050] NET: Registered protocol family 16
[    0.674050] NET: Registered protocol family 16
[    1.133094] usbcore: registered new interface driver usbfs
[    1.133094] usbcore: registered new interface driver usbfs
[    1.144299] usbcore: registered new interface driver hub
[    1.144299] usbcore: registered new interface driver hub
[    1.155317] usbcore: registered new device driver usb
[    1.155317] usbcore: registered new device driver usb
[    1.206476] clocksource: Switched to clocksource riscv_clocksource
[    1.206476] clocksource: Switched to clocksource riscv_clocksource
[    1.283324] NET: Registered protocol family 2
[    1.283324] NET: Registered protocol family 2
[    1.323250] tcp_listen_portaddr_hash hash table entries: 512 (order: 1, 8192 bytes)
[    1.323250] tcp_listen_portaddr_hash hash table entries: 512 (order: 1, 8192 bytes)
[    1.337567] TCP established hash table entries: 8192 (order: 4, 65536 bytes)
[    1.337567] TCP established hash table entries: 8192 (order: 4, 65536 bytes)
[    1.353470] TCP bind hash table entries: 8192 (order: 5, 131072 bytes)
[    1.353470] TCP bind hash table entries: 8192 (order: 5, 131072 bytes)
[    1.372896] TCP: Hash tables configured (established 8192 bind 8192)
[    1.372896] TCP: Hash tables configured (established 8192 bind 8192)
[    1.396170] UDP hash table entries: 512 (order: 2, 16384 bytes)
[    1.396170] UDP hash table entries: 512 (order: 2, 16384 bytes)
[    1.406758] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes)
[    1.406758] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes)
[    1.424474] NET: Registered protocol family 1
[    1.424474] NET: Registered protocol family 1
[    1.467000] RPC: Registered named UNIX socket transport module.
[    1.467000] RPC: Registered named UNIX socket transport module.
[    1.476092] RPC: Registered udp transport module.
[    1.476092] RPC: Registered udp transport module.
[    1.483301] RPC: Registered tcp transport module.
[    1.483301] RPC: Registered tcp transport module.
[    1.492051] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    1.492051] RPC: Registered tcp NFSv4.1 backchannel transport module.
[   20.650653] workingset: timestamp_bits=62 max_order=18 bucket_order=0
[   20.650653] workingset: timestamp_bits=62 max_order=18 bucket_order=0
[   21.301737] NFS: Registering the id_resolver key type
[   21.301737] NFS: Registering the id_resolver key type
[   21.310318] Key type id_resolver registered
[   21.310318] Key type id_resolver registered
[   21.316525] Key type id_legacy registered
[   21.316525] Key type id_legacy registered
[   21.323057] nfs4filelayout_init: NFSv4 File Layout Driver Registering...
[   21.323057] nfs4filelayout_init: NFSv4 File Layout Driver Registering...
[   21.812561] io scheduler mq-deadline registered
[   21.812561] io scheduler mq-deadline registered
[   21.820217] io scheduler kyber registered
[   21.820217] io scheduler kyber registered
[   24.987508] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[   24.987508] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[   25.071357] 10000000.uart: ttyS0 at MMIO 0x10000000 (irq = 1, base_baud = 3125000) is a TI16750
[   25.071357] 10000000.uart: ttyS0 at MMIO 0x10000000 (irq = 1, base_baud = 3125000) is a TI16750
[   26.495808] printk: console [ttyS0] enabled
[   26.495808] printk: console [ttyS0] enabled
[   26.575885] xilinx_spi 20000000.xps-spi: at 0x20000000 mapped to 0x(____ptrval____), irq=2
[   26.575885] xilinx_spi 20000000.xps-spi: at 0x20000000 mapped to 0x(____ptrval____), irq=2
[   26.620357] libphy: Fixed MDIO Bus: probed
[   26.620357] libphy: Fixed MDIO Bus: probed
[   26.650677] lowrisc-digilent-ethernet: Lowrisc ethernet platform (30000000-30007FFF) mapped to ffffffd004028000
[   26.650677] lowrisc-digilent-ethernet: Lowrisc ethernet platform (30000000-30007FFF) mapped to ffffffd004028000

zarubaf, 2019-10-22:
Congratulations. 

You can either alter the Linux config:

https://github.com/pulp-platform/ariane-sdk/blob/1ee3c6179fd061b541bd7d8a4583699fcb34f87e/configs/linux_defconfig#L27

or try to delete the device tree entry, re-compile the bootrom and re-run the flow.

RaphaelKlink, 2019-10-23:
Thank you that fixed the problem. The Core is now booting to the login.

And I used the DDR4 Interface of your VCU118 implementation, it is working.

zarubaf, 2019-10-23:
May I ask what’s the clock speed you are achieving? Would you like to contribute your changes to the project?

RaphaelKlink, 2019-10-23:
I have not tried to increase the clock speed.
When I get the the JTAG running, I would be happy.
But the SD Card, the UART and the JTAG are all working via PMOD and external Devices.

RaphaelKlink, 2019-10-30:
I am new to Openocd.
How do I access the gdb connection?

I have tried to use the same JTAG chip as in the genesys2. Is there a better alternative for an external JTAG adapter?
 
fpga@fpgaserverVCU108:~$ lsusb
Bus 003 Device 002: ID 0403:6010 Future Technology Devices International, Ltd FT2232C Dual USB-UART/FIFO IC

fpga@fpgaserverVCU108:~/ariane/fpga$ openocd -f ariane.cfg
Open On-Chip Debugger 0.10.0+dev-00822-g0409bf9b2 (2019-10-22-15:42)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=64, misa=0x800000000014112d
Info : Listening on port 3333 for gdb connections
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

Is the last line missing in the openocd output important ?

RaphaelKlink, 2019-10-30:
I have a problem with the JTAG now.

I tried to use the same JTAG chip as the Genesys2, but it does not work.

There is a Config File for an external JTAG Adapter, is this working and for with Adapter is this File?

This is the output when i tried to access the gdp connection.

Open On-Chip Debugger 0.10.0+dev-00822-g0409bf9b2 (2019-10-22-15:42)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=64, misa=0x800000000014112d
Info : Listening on port 3333 for gdb connections
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : dropped 'gdb' connection

(gdb) target remote localhost:3333
Remote debugging using localhost:3333
Warnung: Architecture rejected target-supplied description
Warnung: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
Truncated register 37 in remote 'g' packet

The Problem is also the chip is detected differently
Bus 003 Device 002: ID 0403:6010 Future Technology Devices International, Ltd FT2232C Dual USB-UART/FIFO IC

closed 2019-11-12 by RaphaelKlink