openhwfoundation/cva6 #342: Syntax error near ' in lzc.sv

voldemoriarty, 2019-10-26:
Running make fpga with Vivado 2017.4 gives me this error in [lzc.sv:90](https://github.com/pulp-platform/common_cells/blob/790f2385c01c83022474eede55809666209216e3/src/lzc.sv#L90)
```  
assign cnt_o   = NUM_LEVELS > unsigned'(0) ? index_nodes[0] : $clog2(WIDTH)'(0);
```
changing the `$clog(WIDTH)'(0)` to `'b0` fixes it and the synth is running now.

My question is is this a valid fix?

voldemoriarty, 2019-10-26:
UPDATE: bitstream generation successful

rahulkande, 2019-10-30:
Hi,
I was getting a similar error. If you look at the latest version of that file (https://github.com/pulp-platform/common_cells/blob/master/src/lzc.sv), they already fixed the issue by replacing it with {($clog2(WIDTH)){1'b0}};
It looks like you were able to resolve the issue, but, just letting you know.

closed 2019-10-30 by voldemoriarty