openhwfoundation/cva6 #345: Ariane- synthesized error for $clog2

georgelee79, 2019-11-06:
Hi, dear pulp platform support members
We download Ariane RTL source code(version 4.2.0 - 2019-06-04), and synthesize it by Synopsys Design Compiler (version 2016.12), but some error happened for $clog2 during DC compiling process.
The detail error information is as below:
The construct "constant function call as casting type except with $bits()" is not supported.(VER-721)

Please help us to check the root cause of this error. 
You can see the error details in the attached files.
Thank you very much.
[image]
[image]

zarubaf, 2019-11-06:
The Ariane (`v4.2`) which you can find online has been taped out several times. I think either you are using an too old version of Synopsys (unlikely) or you didn't activate SystemVerilog support (`analyze -format sv ../path/to/file.sv`).

We use Synopsys for our tape outs as well. Let me know how it goes.

georgelee79, 2019-11-07:
Dear Zaruba, when you synthesize Ariane (v4.2) , which version of Synopsys Design Compiler do you use? thanks

zarubaf, 2019-11-12:
`2017.09`

closed 2020-01-22 by zarubaf