openhwfoundation/cva6 #348: SPIKE memory initialization for GRM

gpersyn, 2019-11-11:
Howdy,

I am using the included SPIKE emulator as a GRM. I have noticed that I am getting mismatches when there is an ld instruction followed by a bnez instruction. I believe this is caused by the SPIKE emulator having its memory initialized to random values while Araine's memory is initialized to zeros. So SPIKE takes the branch while Ariane does not. Does anyone know how to initialize the SPIKE memory to be zeros? 

Mismatch:
------------------------------------------------------------
Write back data mismatches
ERROR: [0000000000000000 === 00007fefef000000 @ PC 000000008000167a]

Ariane trace:
------------------------------------------------------------
000000008000167a 0 00006414 c.ld           a3, 8(s0)             a3  0000000000000000 
000000008000167c 0 0000e681 c.bnez         a3, pc + 8            a3  0000000000000000
000000008000167e 0 0000854e c.mv           a0, x0, s3            a0  0000000000000000 s3  :0000000000000000

Spike trace:
------------------------------------------------------------
0x000000008000167a (0x00006414) c.ld    a3, 8(s0)
0x000000008000167c (0x0000e681) c.bnez  a3, pc + 8
0x0000000080001684 (0x000084ca) c.mv    s1, s2

zarubaf, 2019-11-12:
I actually thought that Spike also initializes its memory to zero. But anyway, isn't the problem more that you are reading from unitialized memory?

gpersyn, 2019-11-14:
For our purposes, reading from uninitialized memory is not a problem. We just want Ariane and SPIKE to handle it the same so there is no mismatch.

closed 2020-01-22 by zarubaf