openhwfoundation/cva6 #349: JTAG problem on VCU 108

RaphaelKlink, 2019-11-12:
I have a problem with the JTAG now.

I tried to use the same JTAG chip as the Genesys2, but it does not work.

There is a Config File for an external JTAG Adapter, is this working and for with Adapter is this File?

This is the output when i tried to access the gdp connection.

Open On-Chip Debugger 0.10.0+dev-00822-g0409bf9b2 (2019-10-22-15:42)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select '.
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (), part: 0x0000, ver: 0x0)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info : hart 0: XLEN=64, misa=0x800000000014112d
Info : Listening on port 3333 for gdb connections
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : dropped 'gdb' connection

(gdb) target remote localhost:3333
Remote debugging using localhost:3333
Warnung: Architecture rejected target-supplied description
Warnung: No executable has been specified and target does not support
determining executable automatically. Try using the "file" command.
Truncated register 37 in remote 'g' packet

The Problem is also the chip is detected differently
Bus 003 Device 002: ID 0403:6010 Future Technology Devices International, Ltd FT2232C Dual USB-UART/FIFO IC

zarubaf, 2020-01-07:
The FTDI chip is specific to the Genesys II board and will only work with this specific board. You will need to check which debug transport you can use with the VCU and whether there is a driver for OpenOCD. It is interesting though that OpenOCD found a tap and seems to have successfully identified the hart. I have never seen that kind of error when connecting with gdb.

closed 2020-01-07 by zarubaf

RaphaelKlink, 2020-01-07:
There are config files for a PMOD JTAG adapter ( e.g. ariane_pmod.cfg) with Olimex Ltd. ARM-USB-OCD-H JTAG+RS232 as ftdi device.

Is this the Device and was this tested? 
https://www.olimex.com/Products/ARM/JTAG/ARM-USB-OCD-H/

How do the other FPGA Boards like VCU707 handle the programming ?

RaphaelKlink, 2020-01-08:
I have found the error

I only started gdb not risc64-unkown-elf-gdb, it could not handle the architecture.
But I can not run any file. When I tried it with the hello.c example i get this output on the UART.
Furthermore Ariane gets stuck completely.
The virtual address is the start address of the program from gdb.

 [  634.751483] Unable to handle kernel paging request at virtual address 00000000000100c6
[  634.751483] Unable to handle kernel paging request at virtual address 00000000000100c6
[  634.770582] Oops [#1]
[  634.770582] Oops [#1]
[  634.783855] Modules linked in:
[  634.783855] Modules linked in:
[  634.793451] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.1.0-rc7 #2
[  634.793451] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.1.0-rc7 #2
[  634.810122] sepc: 00000000000100c6 ra : ffffffe0006074f0 sp : ffffffe000a8bf30
[  634.810122] sepc: 00000000000100c6 ra : ffffffe0006074f0 sp : ffffffe000a8bf30
[  634.834192]  gp : ffffffe000ac1b18 tp : ffffffe000a95cc0 t0 : ffffffe03ada44f0
[  634.834192]  gp : ffffffe000ac1b18 tp : ffffffe000a95cc0 t0 : ffffffe03ada44f0
[  634.859016]  t1 : 0000000000000000 t2 : 0000000000000000 s0 : ffffffe000a8bf40
[  634.859016]  t1 : 0000000000000000 t2 : 0000000000000000 s0 : ffffffe000a8bf40
[  634.883876]  s1 : ffffffe000ac1b88 a0 : ffffffe03af90f40 a1 : 0000000000001000
[  634.883876]  s1 : ffffffe000ac1b88 a0 : ffffffe03af90f40 a1 : 0000000000001000
[  634.908716]  a2 : ffffffe03ab0fdc0 a3 : 4000000000000000 a4 : ffffffe03af91a78
[  634.908716]  a2 : ffffffe03ab0fdc0 a3 : 4000000000000000 a4 : ffffffe03af91a78
[  634.933529]  a5 : 0000000000000000 a6 : 0000000000000000 a7 : 00000000efe4b99a
[  634.933529]  a5 : 0000000000000000 a6 : 0000000000000000 a7 : 00000000efe4b99a
[  634.958393]  s2 : 0000000000000001 s3 : ffffffe000ac1d2c s4 : ffffffe000abca1b
[  634.958393]  s2 : 0000000000000001 s3 : ffffffe000ac1d2c s4 : ffffffe000abca1b
[  634.983246]  s5 : ffffffe000a1af38 s6 : 0000000000000001 s7 : 0000000000000000
[  634.983246]  s5 : ffffffe000a1af38 s6 : 0000000000000001 s7 : 0000000000000000
[  635.008095]  s8 : 0000000000000000 s9 : 0000000000000000 s10: 0000000000000000
[  635.008095]  s8 : 0000000000000000 s9 : 0000000000000000 s10: 0000000000000000
[  635.032965]  s11: 0000000000000000 t3 : 00000000efe4b99a t4 : ffffffe000a89308
[  635.032965]  s11: 0000000000000000 t3 : 00000000efe4b99a t4 : ffffffe000a89308
[  635.057790]  t5 : 0000000000000002 t6 : ffffffe03ab0fa79
[  635.057790]  t5 : 0000000000000002 t6 : ffffffe03ab0fa79
[  635.078131] sstatus: 0000000200000120 sbadaddr: 00000000000100c6 scause: 000000000000000c
[  635.078131] sstatus: 0000000200000120 sbadaddr: 00000000000100c6 scause: 000000000000000c
[  635.112348] ---[ end trace cd7567c9408bd43e ]---
[  635.112348] ---[ end trace cd7567c9408bd43e ]---
[  635.123766] Kernel panic - not syncing: Attempted to kill the idle task!
[  635.123766] Kernel panic - not syncing: Attempted to kill the idle task!
[  635.143486] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]---
[  635.143486] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]---

the GDB output is this:

fpga@fpgaserverVCU108:~$ riscv-gnu-toolchain/bin/riscv64-unknown-elf-gdb ./hello.elf
GNU gdb (GDB) 8.3.0.20190516-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./hello.elf...
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x00000000000100c6 in _start ()
(gdb) load
Loading section .text, size 0x24e2 lma 0x100b0
Loading section .rodata, size 0x18 lma 0x12598
Loading section .eh_frame, size 0x4 lma 0x135b0
Loading section .init_array, size 0x10 lma 0x135b8
Loading section .fini_array, size 0x8 lma 0x135c8
Loading section .data, size 0xf58 lma 0x135d0
Loading section .sdata, size 0x40 lma 0x14528
Start address 0x100c6, load size 13486
Transfer rate: 47 KB/sec, 1926 bytes/write.
(gdb) b main
breakpoint 1 at 0x1016e: file hello.c, line 4.
(gdb) c