openhwfoundation/cva6 #355: About the fence instruction

zhaoxiahust, 2019-11-29:
Hi Guys,
  I noticed the current fence instruction implementation in Ariane flushes the L1 cache. Based on my understanding of the fence in the memory consistency model, it only needs to prevent the re-ordering of load-store instructions. I understand based on the current Ariane implementation, L1 cache needs to be flushed to remove some false data brought by the re-ordered loads.
  
  However, if I implement the fence instruction by preventing the load-store re-ordering, do I still need to flush the L1 cache or not? Based on the explanation of the fence instruction in the RISC-V manual, my implementation does not need to flush L1. However, I am afraid Ariane implements fence instruction for other reasons except for maintaining the memory-ordering as specified by the RISC-V manual. Any comments about this?

  Thanks for your help!

Cheers,

hyf6661669, 2019-11-30:
To my understanding, if you only have 1 core in your SoC, you don't have to invalidate the whole dcache for fence instruction. If you have multi cores in your SoC, it's might be needed.
https://github.com/riscv/riscv-isa-manual/issues/252

zhaoxiahust, 2019-11-30:
Based on my understanding, fence instruction is related to memory consistency. About the cache coherence support, we have the snooping or directory protocol which is another story::)))

I am just not sure whether the Ariane's authors use the fence instruction to flush the L1 cache in order to customize the Linux kernel or BBL.

zarubaf, 2020-01-24:
For the single core Ariane system (as you can find in this repository) we originally supported two kinds of cache: a write-through and a write-back one (only write-through is maintained at the moment). All without providing a hardware cache-coherency protocol.

Informally: The `fence` instruction will make sure that other harts will see the stores the executing core has done and this hart will see updates from the other harts in the system. For the write-back cache, this means invalidating the cache and writing-back dirty values. The write-through cache must flush its write-back buffer and invalidate its cache entries (because potentially other cores could have written the same cache line). You can see the difference between the two caches in `controller.sv`.

That being said both is not really implemented and we rely on the fact that we are the only master in this rather simple system.

But, the same write-through cache can also be used in the multi-core OpenPiton system. There they use a full-blown hardware coherency mechanism. The `fence` instruction will only make sure that the write-buffer is empty.

closed 2020-04-15 by zarubaf