openhwfoundation/cva6 #360: Addressed are defined on 64 bits, which generates kgate overhead

JeanRochCoulon, 2019-12-10:
In RISCV specification, two types of address are defined: physical and virtual address. Physical address is max 56 bits and virtual address is 39bit (sv39) or 48bits (sv48). 

In Ariane implementation, physical and virtual addresses are implemented on 64bits.
Address length dimensions a lot of functionalities, scoreboard or caches for instance. By implementing two RTL variables, PLEN (as physical length) and VLEN (as virtual length), address length becomes configurable. By setting PLEN=32 and VLEN=39, kgate reduction is about 40kgates.

I think it would be nice to integrate such a modification in main line to stick to the RISCV specification and reduce the gate count.

zarubaf, 2020-01-07:
I absolutely agree. I would be more than happy to discuss a patch for it.

zarubaf, 2020-02-24:
Thanks for your contribution. Has been closed by #369.

closed 2020-02-24 by zarubaf