openhwfoundation/cva6 #364: ASID - missing features

suppamax, 2019-12-19:
Hi everyone,

in case I'd like to use ASID_WIDTH>1, which feature should I implement?
in an earlier answer (https://github.com/pulp-platform/ariane/issues/291) you stated "nor is the selective flushing logic implemented", but then I don't understand what is the purpose of tlb.sv line 104.

Regards,
Max

zarubaf, 2020-01-07:
The ASID selective flushing logic is there but the supplied ASID is wrong, this should come from the ASID supplied by the `sfence.vma`.

marcosabatano, 2020-01-14:
Hello Zarubaf, 

I am collaborating with suppamax in implementing the ASIDs support (16 bits) in Ariane.
I am relatively new of both Ariane and RISCV, so I spent the last few days trying to build up some understanding of the issue. 

To me It looks like Ariane is pretty well prepared for ASIDs. I found the following issues/missing points: 

1) As you said, the current implementation is using the ASID provided by the SATP register for the flush. This is wrong since we want to be able to flush other ASIDs from the current address space, therefore we should change this taking the "ASID_TO_BE_FLUSHED" from sfence.vma rs2 argument. 

2) The "Global mapping" bit is currently not used in the flush nor in the translation (lookup), that is also missing/wrong in my opinion because: 
   2A) A flush with rs2!=0, should not touch the entries with the "G" flag set.
          This is probably not critical but is impacting performances.  
   2B) During lookups we don't need to match ASIDs in entries with flag "G" set. (is that correct?). 

side question about 2B, what ASID should be set for a "Global address"?  should be 0 or it actually don't care? 

Can you please confirm that what I found make senses (mostly the point 2) ? 
BR
Marco

zarubaf, 2020-01-22:
Hi Marco,

thanks for getting in contact.

1. Yes exactly. Probably the easiest would be to make the `sfence.vma` instruction read `rs2`. Then in the commit stage, when decoding an `sfence.vma` instruction `rs2` can be read from `commit_instr_i` and propagated to the TLBs.

2. I think its still semantically correct to always enforce ordering, but as you said this might cost some performance. I think it doesn't matter what software sets for ASID for a global mapping, hardware should not match the ASID when `G` is set (at least that's how I understood it). Yes, if `rs2 != 0` entries with `G` set, are not ordered (which does not mean that they CAN be ordered. 

If in doubt you can also ask on the official RISC-V mailing list. People are usually helpfull there ;-) Hope that helps.

marcosabatano, 2020-01-23:
Hi zarubaf, 

Thanks a lot for your answer. 
I completed a first implementation of both 1 and 2 (16 bits ASID). I actually store the rs1 and rs2 of a coming SFENCE.VMA instruction in a register I defined in the execution stage, then the HW will use it in the next clock cycle when the "flush" signal gets triggered from the commit stage, in this way the  the critical path it should not be increased in any way. 

We will start testing soon and we will be able to share the implementation once we are confident that everything works how it should. 

BR
Marco

zarubaf, 2020-01-23:
I had some similar thinking in that direction. You just need to be careful in case another `sfence.vma` comes along so that you don't accidentally override the parameters. But I agree my proposed solution _potentially_ has implications on the critical path. 

Perfect, thanks for the heads up. Looking forward to seeing your solution :-)

marcosabatano, 2020-01-24:
As far as I can see, when the SFENCE.VMA is committed, the complete pipeline is flushed. if this is true I only need to ensure that another SFENCE.VMA in the "previous stage" of the pipe cannot overwrite the register until the commit is not completely executed.

Since the pipe is flushed, the next SFENCE.VMA in the code will restart to be executed from IF stage, that allow all the time that i can make that register writable again without synchronization problems. 

in a nutshell: 
After the HW in the Ex stage will recognize a SFENCE.VMA i will make the register not writable until after the execution of the SFENCE. 

BR
Marco

zarubaf, 2020-01-24:
Yes that sounds perfect.

marcosabatano, 2020-04-07:
Hello Florian, 

We have completed and validated our 16 bit ASID implementation in Ariane. 
We would like to share it with you. 

We are using the version 4.2 of Ariane, it's not clear to me how we can share the modification. 
Can you propose a way of proceeding?
My fear is that Ariane changed too much "around" my implementation. 

BR
Marco

zarubaf, 2020-04-07:
That sounds fantastic. How much did you change? I think we probably need to bite the bullet and re-base on the current master. I am a bit short on time at the moment but worst-case I can try re-basing your changes as well. Let me know, and thanks for your contribution!

marcosabatano, 2020-04-07:
I modified the following files: 

include/ariane_pkg.sv
src/ariane.sv
src/commit_stage.sv
src/csr_regfile.sv
src/decoder.sv
src/ex_stage.sv
src/issue_read_operands.sv
src/issue_stage.sv
src/load_store_unit.sv
src/mmu.sv
src/tlb.sv

here attached the modified file, Please let me know if you have any comment and/or you want me to do something else!
BR
Marco
[ariane_4.2_modified_files.tar.gz](https://github.com/pulp-platform/ariane/files/4445974/ariane_4.2_modified_files.tar.gz)

HensoldtCyberHWGroup, 2020-06-21:
Hi Florian, 

About this topic, as you can see from the GitHub messages I just pushed a formal pull request based on the latest Ariane commit (Jun.2020), so It should be straightforward to review and integrate. 

Let me know if you need some resources for the testing. We could provide some functional testing, but those probably needs to be integrated in your environment then.
BR
Marco

MikeOpenHWGroup, 2023-02-18:
Hi @zarubaf, it _looks_ like this issue has been resolved and can be closed.  Can you confirm that?

closed 2023-02-18 by suppamax