openhwfoundation/cva6 #366: Formal Verification & AXI lock query

ZipCPU, 2020-01-16:
I was running Symbiotic's EDA Suite on your [axi_adapter](/src/axi_adapter.sv), and noticed that the AxLOCK flag was held low.

Does this mean that Ariane doesn't support atomic accesses over AXI?

ZipCPU, 2020-01-16:
You mean I checked for compliance at the wrong level?

Doh!

Where would you suggest the right level to check would be?  The location where the AXI request is first fully formed?

As for issues, the biggest issues I came across so far were due to what looked like parameter mismatches.  For example, `DATA_WIDTH` can never be less than 64, but there's also a bug if it is == 64.  Similarly, the ID width in the parameters is set to 10, but internally the interfaces can only ever handle an ID width of 4.  Finally, the CACHELINE_BUS_OFFSET doesn't match the data width.  It should be 5 if the DATA_WIDTH is 256, not 8.

fabianschuiki, 2020-01-27:
I think the AXI spec states that the `AxLOCK` signal is *not* supported in AXI4, but is a remnant of AXI3 that must be kept in place (see "A7.3 Locked accesses" in "AMBA AXI and ACE Protocol Specification; AXI3, AXI4, and AXI4-Lite; ACE and ACE-Lite"). We do this but tie the signal low. All locking operations are handled via atomics/`AxATOP`.

ZipCPU, 2020-01-27:
You might wish to read the AXI4 specification again.  AXI3 allows "locking" the bus, requiring 2-bit AxLOCK flags to do so.  AXI4 has a different mechanism for achieving atomic access.  It still uses the AxLOCK flag--but only 1'b1 of what was the AXI3 AxLOCK bits.  If set in AXI4, AxLOCK indicates an exclusive access burst.  Exclusive access bursts have particular semantics, described in section A.7.2 of the spec.

At issue is whether or not your atomic access mechanism is specification compliant.  If no other master exists on the bus, then I suppose you could bastardize the AXI protocol in order to handle atomics some other way.  This would not be a specification compliant implementation.  On the other hand, if another master shares the bus with the CPU, then the AxLOCK flag is the only specification compliant mechanism for supporting atomic access via exclusive access bursts.  See section A.7.2 of the same specification for the semantics of such an exclusive access transaction.

zarubaf, 2020-01-27:
This is where the AXI request is formed https://github.com/pulp-platform/ariane/blob/master/src/cache_subsystem/wt_axi_adapter.sv and https://github.com/pulp-platform/ariane/blob/48be94f822884e33a80c7226d6f0489ae8235de4/src/axi_shim.sv As you can see `AxLOCK` is set there.

closed 2020-02-03 by zarubaf