openhwfoundation/cva6 #370: dcache tries to cache I/O memory

jrrk2, 2020-01-30:
In normal operation Ariane only generates access to I/O that is explicitly programmed in the device driver. But for some reason the dcache is trying to read these addresses as well, resulting in multiple accesses on the peripheral bus and consequent non-idempotency.

[image]

jrrk2, 2020-01-31:
This unfortunate behaviour seems to arise because the if clause at line wt_dcache_missunit.sv:411 is triggered for non-cacheable addresses. I don't understand what the implications of trying to fix this would be.

`+++ b/src/cache_subsystem/wt_dcache_missunit.sv
@@ -408,7 +408,7 @@ module wt_dcache_missunit #(
             if (mshr_rdrd_collision_d[miss_port_idx]) begin
               miss_replay_o[miss_port_idx] = 1'b1;
             // stall in case this CL address overlaps with a write TX that is in flight
-            end else if (!tx_rdwr_collision) begin
+            end else if (some_guard_expr && !tx_rdwr_collision) begin // JRRK hack to test whether no-cache mem is cached
`

zarubaf, 2020-02-03:
Thanks for digging this up. After a lengthy discussion with Michael we are pretty sure, that given the fix I've provided in #368 (given that it is correct) should prevent such features. Nevertheless, we found a potential issue where load/store ordering to the peripheral space could be violated.

I understand what you are saying with the issue but I am not quite sure whether I completely understand the waves and trace you have posted. Would you mind:
- Giving me access to the wave database.
- The instruction trace in question (maybe incl. the binary).

Then I can take a look myself. I think that would greatly speed-up the debugging process and we can try to get this out of our way. Thanks!

jrrk, 2020-02-03:
Hi Florian, the binary is checked in along with everything you need (apart from vcs and Vivado) to reproduce the issue. I would suggest you try it with and without my suggested patch, and you will soon see the out of order I/O happening. Essentially the miss unit tries to grab new data without draining the write buffer first. If you can’t reproduce the issue, maybe we should talk via Skype. Further info: with just your patch, Linux does not boot at all (blank screen), with your patch and my patch, Linux boots but goes to systemd failed mode. So we need to find at least one more fix to the fix.

zarubaf, 2020-02-03:
Thanks. I could successfully compile your version (https://github.com/jrrk2/ariane/tree/issue-356) with VCS. 

- Can you tell me how to exactly reproduce the issue (e.g.: how to call `simv`). `./simv -nogui -l simv.log` gives me a long log of UART displays. 
- How can I dump the waves in VCS?
- Where is the binary checked-in? To get the objdump.

Thanks. We can also arrange for a Skype call if you prefer.

jrrk2, 2020-02-03:
If you compile with -gui then you will get a waveform display unless you execute with -nogui.
However you will not see a problem because this is the version with my fix. Please refer to [testcase](https://github.com/pulp-platform/ariane/pull/372), for your version with the revised software and peripherals. I have added the instruction to dump all signals to vcd (as well as uploading my waveform display in .fst for gtkwave use, for comparison with yours)

zarubaf, 2020-02-24:
Closed by #368

closed 2020-02-24 by zarubaf