openhwfoundation/cva6 #383: Add Chipyard description to README

abejgonzalez, 2020-02-26:
Related Issues/PRs: #382, https://github.com/ucb-bar/chipyard/pull/448

Once Chipyard is fully updated with Ariane support, the README should be updated to point to Chipyard/Chipyard Documentation on how to use the core there.

gabaxter, 2021-02-11:
Has anyone actually used a CVA6 core within chipyard?  I can find no documented examples, nor any use cases.  While I understand that Chipyard is incapable of producing Verilator model, docs say it can produce a VCS model.  What I need fundamentally is an example of how to get to at least VCS for CVA6.  

Why is it that CVA6 using verilog source is not supported in Verilator?  If chipyard supports allowed external modules of Verilog (e.g. AXI IP) to be included, why can it not produce for the CVA6 tile which has been matched to rocket with an AXI interface?  For one CVA6 core, this seems it ought to be able to work.  Can someone define why it will not?

abejgonzalez, 2021-02-11:
Firstly, this question seems more oriented to the Chipyard mailing list than here: https://groups.google.com/forum/#!forum/chipyard

To directly answer your question. Yes, I've used (an older) version of CVA6 in Chipyard and was able to successfully boot Linux in FireSim simulations on it (others have used it too https://github.com/ucb-bar/chipyard/issues/759). We have a configuration for it that can be used like any other Chipyard config (with Rocket/BOOM/Sodor) so all you have to do is follow the instructions for the other cores (since all the infrastructure is shared between them). Basically in `sims/vcs` do `make CONFIG=CVA6Config` (see https://chipyard.readthedocs.io/en/latest/Simulation/Software-RTL-Simulation.html). If the docs are unclear on how we organize things, configure things, etc please let us know so we can try to be clearer!

The CVA6 version that we have in Chipyard is a bit old at this point so we would love someone to bump it and make sure it works properly with the newest version of Linux that we are running.

As for Verilator support for CVA6 in Chipyard, I think that is a solved issue now (I think those issues were solved when NVDLA was integrated). Tested that specific config in Chipyard 1.4.0 today and it worked. So the docs need to be updated!

gabaxter, 2021-02-11:
abejgonzalez, Thank you very much for the answer and for helping to get farther down the road.  The reason this was posted here was because this issue already existed, and because in my view it has not been addressed. I am finding in both this repo and in the chipyard repo that the Docs are "almost" what is needed.  For example, wanting to build a CVA6 in Chipyard - there are no directions for how to do so.  It is assumed you can go through the myriad of configs present in the repo and build things (if you know what those configs even are).  The docs are very weak in both this repo and chipyard and could stand a lot of improvement.  Something that would be useful is for someone to go through the docs for install and initial use and ensure that every single step is written down, works and is tested (like from a completely clean VM).  That is what makes these docs harder to use (in this repo and the chipyard repo).   If we expect community adoption we need to lower the impedance to use. Docs are the single best use of time for those who are very familiar with all of this to help those of us who are not getting up to speed faster.

Thank you for the 'secret' sauce of `CONFIG=CVA6Config`.  I just tried that in verilator and it appears to have built a sim model, which is outstanding.  That is **very** helpful.  May I recommend that you add that CONFIG to the the chipyard docs in the sim section, along with any other relevant CONFIGs you know.  People can then add CONFIGs and document in brief what the offer and where their source is located?  That will enable new people to find what they are looking for and quickly get onto at least Verilator/VCS.

MikeOpenHWGroup, 2023-02-18:
Hi, this issue is three years old and has received no comments for two years.  AFAIK, CVA6 has no plans to provide explicit support for Chipyard, although we would consider a contribution from the open source community to provide that.

@abejgonzalez and @gabaxter, if you would like to make such a contribution please either open a new issue or pull-request.  Thanks!

closed 2023-02-18 by MikeOpenHWGroup